arkypita/LaserGRBL #729: LaserGRBL and Fusion 360 CAM files

khturnings, 2019-09-10:
Hi
I have been trying to load a simple Gcode file generated by Fusion 360. However it seems that the use of M4 (dynamic Laser) means that LaserGRBL burns transition moves instead of leaving them blank.
Can anyone help with some advice?

Heres the output from Fusion360

%
(test)
G90 G94
G17
G21

(Text 2)
G54
G0 S90 M4
G0 X41.326 Y6.951
G1 F1000
G1 X40.997 Y6.57 F1500
G0 X40.998
G3 X40.558 Y5.361 Z0 I1.378 J-1.186 F1500
G0 X40.205 Y5.716
G3 X40.205 Y4.603 I2.974 J-0.556 F1500
G3 X41.283 Y4.081 I0.732 J0.137
G0 X40.558 Y5.361
G1 Y4.679 F1500
G3 X41.042 Y4.168 I0.511 J0
G1 X41.54 Y4.215
G0 X41.283 Y4.081
G1 X42.278 Y4.603 F1500
G0 X43.532 Y7.253
G1 X42.958 Y4.102 F1500
G3 X43.288 Y3.949 I0.239 J0.083
G3 X44.023 Y4.382 I-0.92 J2.402
G0 X43.674 Y4.135
G2 X43.358 Y4.402 I-0.081 J0.225 F1500
G1 X43.932 Y7.553
G0 X44.023 Y4.382
G1 X44.355 Y4.661 F1500
G0 X45.332 Y7.253
G1 X44.758 Y4.102 F1500
G3 X45.088 Y3.949 I0.239 J0.083
G3 X45.823 Y4.382 I-0.92 J2.402
G0 X45.474 Y4.135
G2 X45.158 Y4.402 I-0.081 J0.225 F1500
G1 X45.732 Y7.553
G0 X46.155 Y4.661
G1 X45.823 Y4.382 F1500
G0 X46.314 Y4.236
G1 X46.642 Y4.135 F1500
G3 X47.539 Y4.265 I0.312 J1.01
G1 X47.648 Y4.502
G3 X47.593 Y4.92 I-0.36 J0.165
G1 X46.816 Y6.004
G0 X47.648 Y4.348
G3 X47.905 Y5.181 I-0.544 J0.625 F1500
G1 X47.172 Y6.23
G2 X47.23 Y6.887 I0.421 J0.294
G1 X47.236 Y6.894
G1 X47.246 Y6.902
G1 X47.27 Y6.922
G1 X47.294 Y6.942
G0 X46.816 Y6.004
G2 X46.816 Y6.435 I0.3 J0.215 F1500
G2 X47.227 Y6.886 I2.119 J-1.519
G1 X47.236 Y6.894
G1 X47.238 Y6.895
G0 X47.317 Y8.723
G3 X46.168 Y8.607 I-0.478 J-1.01 F1500
G3 X45.332 Y7.253 I1.309 J-1.744
G0 X45.732 Y7.553
G2 X46.545 Y8.791 I1.866 J-0.34 F1500
G0 X48.302 Y6.737
G3 X47.539 Y7.035 I-0.669 J-0.587 F1500
G3 X47.294 Y6.942 I0.054 J-0.511
G0 X45.517 Y8.723
G3 X44.368 Y8.607 I-0.478 J-1.01 F1500
G3 X43.532 Y7.253 I1.309 J-1.744
G0 X43.932 Y7.553
G2 X44.744 Y8.791 I1.866 J-0.34 F1500
G0 X41.536 Y6.997
G2 X42.261 Y6.33 I0.092 J-0.627 F1500
G2 X41.368 Y5.423 I-0.968 J0.06
G1 X40.558 Y5.361
G0 X41.216 Y5.412
G1 X41.598 Y5.718 F1500
G3 X41.781 Y6.739 I-0.494 J0.616
G3 X41.326 Y6.951 I-0.384 J-0.23
G0 X39.497 Y6.842
G3 X38.636 Y6.997 I-0.661 J-1.203 F1500
G0 X37.658 Y5.361
G2 X38.097 Y6.57 I1.817 J0.023 F1500
G1 X38.426 Y6.951
G0 X37.305 Y5.716
G3 X37.305 Y4.603 I2.974 J-0.556 F1500
G3 X38.383 Y4.081 I0.732 J0.137
G0 X37.658 Y5.361
G1 Y4.679 F1500
G3 X38.142 Y4.168 I0.511 J0
G1 X38.639 Y4.215
G0 X39.378 Y4.603
G1 X38.383 Y4.081 F1500
G0 X34.777 Y3.993
G3 X36.045 Y4.649 I-0.106 J1.758 F1500
G3 X36.753 Y6.293 I-2.624 J2.105
G3 X36.5 Y7.697 I-2.275 J0.315
G3 X35.9 Y8.137 I-0.763 J-0.411
G0 X35.425 Y8.228
G2 X35.954 Y7.833 I-0.189 J-0.805 F1500
G2 X36.243 Y6.343 I-2.12 J-1.184
G2 X35.405 Y4.285 I-3.251 J0.125
G2 X34.759 Y3.993 I-0.741 J0.776
G2 X34.094 Y4.392 I0.008 J0.766
G2 X33.798 Y5.88 I2.067 J1.186
G2 X34.628 Y7.939 I3.256 J-0.116
G2 X35.274 Y8.239 I0.754 J-0.777
G1 X35.425 Y8.228
G1 X35.9 Y8.137
G0 X35.252 Y8.238
G3 X33.61 Y7.177 I0.001 J-1.802 F1500
G3 X33.316 Y5.185 I3.354 J-1.514
G1 X33.555 Y4.734
G1 X33.878 Y4.33
G1 X34.345 Y4.062
G1 X34.607 Y4.006
G0 X30.802 Y6.737
G3 X30.039 Y7.035 I-0.669 J-0.587 F1500
G3 X29.794 Y6.942 I0.054 J-0.511
G1 X29.746 Y6.902
G1 X29.736 Y6.894
G1 X29.73 Y6.887
G3 X29.672 Y6.23 I0.364 J-0.363
G0 X29.316 Y6.004
G2 X29.316 Y6.435 I0.3 J0.215 F1500
G2 X29.727 Y6.886 I2.119 J-1.519
G1 X29.736 Y6.894
G1 X29.738 Y6.895
G0 X29.817 Y8.723
G3 X28.668 Y8.607 I-0.478 J-1.01 F1500
G3 X27.832 Y7.253 I1.309 J-1.744
G0 X28.232 Y7.553
G2 X29.045 Y8.791 I1.866 J-0.34 F1500
G0 X28.655 Y4.661
G1 X28.323 Y4.382 F1500
G0 X28.815 Y4.236
G1 X29.142 Y4.135 F1500
G3 X30.039 Y4.265 I0.312 J1.01
G1 X30.148 Y4.502
G3 X30.093 Y4.92 I-0.36 J0.165
G1 X29.316 Y6.004
G0 X29.672 Y6.23
G1 X30.405 Y5.181 F1500
G2 X30.148 Y4.348 I-0.801 J-0.209
G0 X27.974 Y4.135
G2 X27.658 Y4.402 I-0.081 J0.225 F1500
G1 X28.232 Y7.553
G0 X28.323 Y4.382
G2 X27.588 Y3.949 I-1.655 J1.969 F1500
G2 X27.258 Y4.102 I-0.091 J0.237
G1 X27.832 Y7.253
G0 X26.55 Y4.624
G1 X25.555 Y4.101 F1500
G0 X25.811 Y4.236
G1 X25.314 Y4.189 F1500
G2 X24.83 Y4.7 I0.027 J0.511
G1 Y5.381
G1 X25.639 Y5.444
G3 X26.533 Y6.351 I-0.075 J0.967
G3 X25.808 Y7.017 I-0.633 J0.039
G3 X24.476 Y5.736 I0.232 J-1.573
G3 X24.476 Y4.624 I2.974 J-0.556
G3 X25.555 Y4.101 I0.732 J0.137
G0 X25.488 Y5.432
G1 X25.87 Y5.739 F1500
G3 X26.052 Y6.76 I-0.494 J0.616
G3 X25.598 Y6.972 I-0.384 J-0.23
G1 X25.269 Y6.59
G3 X24.83 Y5.381 I1.378 J-1.186
G0 X24.094 Y4.774
G1 X23.491 Y4.18 F1500
G2 X22.712 Y4.18 I-0.389 J0.395
G1 X22.677 Y4.213
G1 X22.652 Y4.255
G1 X22.64 Y4.301
G1 X22.642 Y4.35
G0 X22.924 Y5.882
G1 X22.218 Y4.715 F1500
G1 X22.177 Y4.652
G1 X22.133 Y4.591
G0 X22.393 Y7.074
G1 X22 Y6.766 F1500
G3 X21.62 Y6.197 I0.684 J-0.87
G1 X21.385 Y5.37
G3 X21.31 Y4.591 I2.099 J-0.595
G3 X21.707 Y4.345 I0.29 J0.025
G1 X22.133 Y4.591
G2 X21.238 Y4.062 I-1.101 J0.842
G2 X20.839 Y4.449 I-0.052 J0.346
G1 Y4.916
G2 X22.278 Y7.036 I2.28 J0
G2 X23.135 Y7.036 I0.429 J-1.081
G0 X23.205 Y7.414
G1 X22.642 Y4.35 F1500
G0 X23.57 Y4.258
G2 X23.102 Y4.576 I-0.167 J0.258 F1500
G1 X23.781 Y8.045
G2 X23.984 Y8.388 I0.811 J-0.247
G1 X24.306 Y8.721
G1 S0
M30
%

IanJCoombs, 2019-10-01:
I had the same problem.

I think the issue is that the general grbl post processor is not right for lasers. If you go to the [autodesk post processor library](https://cam.autodesk.com/hsmposts) and search for "grbl laser" there is a post processor that works for me.

Add that file to the "Configuration" Folder (You can get there from the post processor dialogue) and then you will see "Grbl Laser" near the existing "Grbl" configuration in the configuration drop down.

khturnings, 2019-10-01:
Hi @IanJCoombs 
I am using the grbl laser post processor, it did not let me use the standard one.

However I believe that I might has resolved the issue, although not tested.

I had not set the GRBL Lasermode ($32) on - https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode.

So files generated by LaserGRBL use 'S' and 'M3' to adjust the power of the laser and turns it off (i.e. S0) before sending the G0 command, I believe this works even if Lasermode is off.
But Fusion360 use Dynamic mode M4 which should mean that G0 does not power on the laser.

I could be wrong but this is my next tests, if I ever get time.
Thanks anyway.

closed 2020-01-27 by arkypita