arkypita/LaserGRBL #731: Step slider does not display

Stevehyman, 2019-09-11:
I have Lasergrbl on another computer, but when I load into this computer the STEP slider does not show on the screen I have tried several down loads and it never shows up. I am using win 8.1, what is going on?
THanks

StuartB4, 2019-09-12:
Check Continuous is not enabled.
[image]

Stevehyman, 2019-09-12:
Thanks, I did that but it still does not show up, but thanks to your reply I think I know what it is, I am using another version of GRBL, that says it's for lasers I will load GRBL1.1 and see if that fixes it.

Stevehyman, 2019-09-12:
Thank, I loaded GRBL1.1 and unchecked Continuos Jog and now every thing is working as it should. Thanks for the wake up.

closed 2020-01-15 by arkypita