arkypita/LaserGRBL #737: mnr pieter pretorius

pieter65, 2019-09-23:
good day can you please tell me if I can change the point of O on the GRBL program I have 3 images that I want to laser on one board. example 1 photo on the left site writing in the middle and foto on the right site can you please help I wish there was a course that I can follow to learn. the ones on YouTube is not useful

StuartB4, 2019-09-23:
Do you mean like this? This was 3 seperate images loaded one after the other using Append File on the file menu.

[image]

pieter65, 2019-09-25:
Thank you for your helpSent from my Huawei phone-------- Original Message --------Subject: Re: [arkypita/LaserGRBL] mnr pieter pretorius (#737)From: Stuart To: arkypita/LaserGRBL CC: pieter65 ,Author Do you mean like this? This was 3 seperate images loaded one after the other using Append File on the file menu.

pieter65, 2019-09-25:
Baie dankie waardeerSent from my Huawei phone-------- Original Message --------Subject: Re: [arkypita/LaserGRBL] mnr pieter pretorius (#737)From: Stuart To: arkypita/LaserGRBL CC: pieter65 ,Author Do you mean like this? This was 3 seperate images loaded one after the other using Append File on the file menu.

closed 2020-01-15 by arkypita