arkypita/LaserGRBL #745: Laser works only in pulse mode (dot mode)

kuksi021, 2019-10-08:
I successfully engraved one picture. After that I imported other picture on some way but laser started engraving process in "dot mode". I cannot change this mode.
I tried to burn some vector but laser only work in same mode. 
[Problem with Software.pdf](https://github.com/arkypita/LaserGRBL/files/3704024/Problem.with.Software.pdf)

Can you tell me what can be a problem.

StuartB4, 2019-10-08:
A couple of things look wrong to me. 
In the speed settings in Target Image you have it set to 10000. That seems way to fast.
Also in Target image settings the laser power setting for S-Max has a red box around it.
Thats because you have it set to 1000 and the setting in GRBL is probably set to 255.
Have a look in Grbl menu, Grbl configuration and see what the setting is for $30
[image]

Also if you have Laser mode set to 1 ($32=1) then change Laser On setting in Target Image settings to M4 instead of M3.

StuartB4, 2019-10-09:
Hi kuksi021. Did you have any luck with it?

closed 2020-01-15 by arkypita