arkypita/LaserGRBL #747: Anel1210

Anel1210, 2019-10-18:
Hi.  I have nothing to zero knowledge about technology, over and above the way I am showed (mostly be youtube video) how to do something.  I have recently bought the GK4545Pro Laser Engraver/Cutter and with it i was told to install the LaserGBRL software.  What a mission it was, but needless to say, I managed to get it done after 2 weeks.  I now only have a problem with the sizing of the image.  When i select for example 100mmx100mm, the actual image does not turn out to be the same size as what i have selected.  So when i want to do a specific size, i am either to large for my material or too small.  I need your assistance, if possible in a video how to set the sizing so that it would be correct and accurate (or as accurate as possible).  then I would also like to know if there is a way that you can preview your image before you start?  And also if there is a way where you can select your size for the X and the Y without the program keeping it in aspect ratio.  Please help this dummy (i think i am the queen of the dummy books).

StuartB4, 2019-10-18:
Hi.
Usually when you load an image in to LaserGRBL you prepare it for engraving by setting things like brightness, contrast and the burning method either Line To Line, 1Bit BW Dithering, Vector or Centerline.
Like in this image:
[image]

Then you select Next and you should get this screen. What ever size you type in the W box it will resize the image but keep the aspect ratio correct, so it will change the numbers in the H box correctly.
**Make sure you  Press Enter after changing the numbers.**
So if your original image was 200 x 200 and you resize it to 70 in the W box it will change the H box to 70 as well. 
[image]

Then select the Engraving Speed at the top and and any other settings you need to change and click Create. That should give you the size you want.

You can preview where the image will be on the material your are engraving.
You need to make a custom button. 
Right click on an empty part of the bar at the bottom of the main window and click Add Custom Button.
[image]

When the window pops up copy and paste this code in to the GCode box:

M3 
S2
F900
G0 X[left] Y[bottom] 
G1 X[right]
G1 Y[top]
G1 X[left] 
G1 Y[bottom] 
G0 X0 Y0 F4000
M5

Then enter a Tooltip (anything you want) in the next box down and then select Idle Program from the drop down selector underneath that and click Save. You should now have an extra button on the main window bar at the bottom. 
To use the button you need to be connected, then select your image and prepare it like earlier.
When you've clicked Create you should be able to press the Custom button you made and it should draw a box around where it is going to engrave, it will put the laser on in low power mode so you can see where it goes and position your material. Then it will switch the laser off and return to the home position ready to engrave.

StuartB4, 2019-10-18:
>  And also if there is a way where you can select your size for the X and the Y without the program keeping it in aspect ratio. 

You would need to do that in a graphics program first. If it did that it would distort your image, it always retains the aspect ratio.

What you can do is select the size of the image by drawing a box around it in Target Image settings where you adjust everything else. Like this:

[image]

Anel1210, 2019-10-19:
Hi there

Thank you. I will do the ekstra button.

1. My problem with the sizing is that for example I selected 100x100mm (my
image is square). When it is finished doing the engraving the actual image
came out 40x40mm and not 100x100mm as I selected. I went back to the image
in the window as you attached and it still showed 100x100mm as I selected.
So then I went and checked the Gbrl configuration settings as the
manufacturer of the machine sent them to me. I changed everything to match
what they sent me and tested it again. It still did the same as before.

So I think either they missed something or their settings is not correct. I
am not sure.

So maybe you could assist with the sizing problem and the correct settings
that I must have. I am attaching the settings they gave to me for you to
check.

Regards
Anel

StuartB4, 2019-10-19:
I can't see you settings anywhere, did you attach them?

StuartB4, 2019-10-19:
It sounds like you steps / mm for the stepper motors are incorrect.

Anel1210, 2019-10-19:
Yes I did attach them. Attaching again now

Anel1210, 2019-10-19:
Did you find my attachment?

Anel1210, 2019-10-19:
So how do I fix it?

StuartB4, 2019-10-19:
There's still nothing there, what are you attaching, what type of file? Some people have problems when trying to attach things from their phone.

Anel1210, 2019-10-19:
Yes it is from my phone. Do you have a number where I can send via
whatsapp?

StuartB4, 2019-10-19:
I don't use whatsapp.

Anel1210, 2019-10-19:
OK. I don't know another way to forward the photo they gave me with the
settings. I only have whatsapp and Facebook. Other then that I don't use
other social media apps. In South Africa we are not very advanced when it
comes to technology.

Anel1210, 2019-10-19:
I took a screenshot of the actual jpeg image. I am attaching and hope it
will work.

StuartB4, 2019-10-19:
You can upload it here:
https://imgbb.com
Then you get a link to send to me and I can view it.
Just copy and paste the link you get in a reply here.

Anel1210, 2019-10-19:
https://ibb.co/StdKBsB

StuartB4, 2019-10-19:
Thats better, I can see it now.
What you need to try is changing $100 and $101.

Start LaserGRBL and click Connect.
Then go to the Grbl menu top left. Click Grbl Configuration.
You should have a window the same as the picture you uploaded.

[image]

To test it just use the same image after trying different settings.

StuartB4, 2019-10-19:
Try 250.

Anel1210, 2019-10-20:
Hi Stuart.

OK I will try and let you know. So I take it that this settings are not the
same for everyone. Why would this be and what is determining that value?
You said I must try 250. If it still isn't correct must it go up or down
and what is the maximum and minimum value it can go? This seems like Greek
to me. Obviously something I don't understand at all. Bearing in mind that
I am not very good with technology.

StuartB4, 2019-10-20:
A lot of them are different because there are a lot of different stepper motors that people use.
If 250 is not correct try a lower number.
What you can do is make mark with a pencil on the X axis rail when the laser is all the way across to the left. Then jog the X Axis 100 mm with the right jog button in lasergrbl, then when it stops make another mark and measure the distance between the marks, to see how much it actually moved.
Then let me know the number and what the setting number is set to for $100.

Like this:

[image]

[image]

[image]

Anel1210, 2019-10-20:
Hi Stuart.

OK. I am going to upload the photos I took to ensure I have done it
correctly on that link you sent me.

123mm distance between the 2 marks. Setting fir $100 is 100.000

Hope this is done correctly

Anel

Anel1210, 2019-10-20:
https://ibb.co/m9GSFZ4 first image ref no

Anel1210, 2019-10-20:
https://ibb.co/DK8TsHZ 2nd picture code

Anel1210, 2019-10-20:
https://ibb.co/M6jzggV 3rd picture

StuartB4, 2019-10-20:
Ok try setting $100 at 81.3

StuartB4, 2019-10-20:
If that does'nt work try 160

Anel1210, 2019-10-20:
OK. So I must only set the X axis at that figure. I leave the y axis at 100
like it is?

StuartB4, 2019-10-20:
No, change them both to the same.

Anel1210, 2019-10-20:
OK. I will do so

Anel1210, 2019-10-20:
OK I will do so. So whatever figure is set for the one must be for the
other one also.

StuartB4, 2019-10-20:
Yes
.

StuartB4, 2019-10-20:
If you have the same motors for X and Y then yes.

Anel1210, 2019-10-20:
OK will try and let you know. Thank you once again.

StuartB4, 2019-10-20:
Ok.

Anel1210, 2019-10-21:
Hi Stuart.

81.3 seems to be the closest. I made image 50mmx12mm and it came out
50mmx10mm. So not much difference.

Thank you so much. Have a great day

Anel1210, 2019-10-21:
Hi Stuart. Just another few questions please.

1. When designing in Inkscape, do I need to set the correct size and then
fit to page rather to get as close as possible to the size I want?

2. On the brightness, contrast I more or less know what it does but the
black and white and the white option I am not too sure

StuartB4, 2019-10-21:
> 81.3 seems to be the closest. I made image 50mmx12mm and it came out
50mmx10mm. So not much difference.

You should be able to get it better than that by trying different numbers.

The Black and White slider is just that, just 2 colours but you can adjust it for more or less.

The White Clip slider lets adjust the image so that some parts dont get burned. 
LaserGRBL will see anything that turns Pink as White as you move the slider, so it won't burn it.
It's handy for getting rid of things like jpeg artifacts or interference on images that are not 
perfect.

In Inkscape I just set the document properties to the size of the engraving I want in mm's.

Anel1210, 2019-10-21:
Thanks. I noticed that when I make the document properties that of the
image size you want when choosing centerline, it Cutts off parts of the
image.

StuartB4, 2019-10-21:
Not sure about that, never used Centerline setting. Also I don't use Inkscape very often because it's too slow on my PC.

Anel1210, 2019-11-04:
Hi Stuart.

Just an update. My program is working wonderfully. I love it.

Is there any other custom buttons that one can add at the bottom with other
nice functions that you may use in the program?

Await to hear from you.

Regards
Anel

StuartB4, 2019-11-04:
I do have 3 more but you would need to upgrade GRBL to version 1.1 to use them
because they use settings that are not included in the version you have.

--------------------------------------------------
Function: Switch Laser on low power for focussing.
Type: TwoStateButton.
Click On - Focus - Click Off.
-----------------------------------------------------------------------
Function: Find center of prepared image. 
Type: TwoStateButton.
Click On - Click Off.
-----------------------------------------------------------------------
Function: Flash Laser.
Type: PushButton. 
Press and release to flash laser on and off. 
-----------------------------------------------------------------------

Anel1210, 2019-11-04:
Could you send me the link for the upgrade please

StuartB4, 2019-11-04:
It depends on which controller board you have. Some boards are harder to do than others.
There is a way to do it using another Laser program.
It's called T2Laser.

They do a trial version so you don't have to buy it to upgrade GRBL.
I've never done it that way but a lot of people use it just for that reason, then if they like T2Laser they keep it and pay for the reg key, or they just use it as a trial version or uninstall it after the upgrade.

I have the instructions for how to do it but you need to download T2Laser and install it first.

Here it is with a picture of the steps.
---------------------------------------
To upgrade to GRBL 1.1e. You can use the trial version of T2laser.
Available here: https://t2laser.wordpress.com/download
Install T2Laser Trial Version. 

Unplug any power supply, you only need USB connection to upgrade it.

Connect your board to your computer using USB.

1.  Start T2Laser and go to File, Control laser. 

2.  On the new window select the correct Com port on the right and click Connect (Lightning icon).
    
3.  It will tell you what version of GRBL it has at the bottom.

4.  Press Machine at the top left and select Upload Firmware (1.1e *DEFAULT*)

5.  Press Yes to confirm. 

6.  It should then do the upgrade, finish and disconnect your board. 

7.  When it's finished Click Connect to confirm it is using GRBL 1.1e. It should tell you at the bottom again.

8.  If you don't want to keep T2Laser you can uninstall it in Add and Remove Programs in Windows Control Panel.

[image]

StuartB4, 2019-11-04:
You will have to do it at you own risk as I have never had to do it myself and don't know if it will work with you controller board.

Anel1210, 2019-11-05:
Thanks Stuart. It seems like I might run into complications which I don't
have energy for at the moment. It's a pity you can't use those functions on
my current version as the centre option would have been very useful. Thank
you anyway. Have a good day

Regards
Anel

StuartB4, 2019-11-05:
Hi.
Try this and let me know if it works.

[image]

Anel1210, 2019-11-05:
Hi.

I will do so, thank you.

I have tried a large project today. My first one. It was about 1.5 hours
running time. I had to do it in sessions as my machine can only run 45min
at a time. I have noticed that on the vectorised option, the border it is
doing is not perfect as with the small projects. Why would that be? I don't
know if I am doing it correct. When my running time reaches 35min, I select
pause, turn off the machine for 20 min. After 20 minutes, I turned on the
machine again and continued with the job. I did it in 3 sessions like that.
May that be the reason why the border is not lining up perfectly or is
there another way I must do the job in sessions?

StuartB4, 2019-11-05:
Why can you only have the laser on for a short time? How many watts is the laser?
Does it have a cooling fan? 
There is a way to cool the laser in the Settings on the GRBL menu, but you also need to upgrade GRBL to version 1.1 use it.

Anel1210, 2019-11-06:
Hi Stuart.

I have the Gk4545pro. 5500watt. I will send you a picture. The vectorise
option, the border doesn't line up perfectly. Then on the cut settings it
doesn't cut through. I did 15 passes and still not cutting through.

https://ibb.co/4fb185B

https://ibb.co/BnJgT69

StuartB4, 2019-11-06:
You may still need to adjust your X and Y steps settings slightly if things are not lining up properly.
What is it you were trying to cut?

Anel1210, 2019-11-07:
Hi Stuart.

I send you the links to go and view the photos. I tried cutting the outside
circle of the watch

StuartB4, 2019-11-07:
I saw the photos. Thats why I said:

> You may still need to adjust your X and Y steps settings slightly if things are not lining up properly.

What material were you trying to cut, wood, plastic, paper?

StuartB4, 2019-11-07:
Also. Have you checked your machine, it could be loosing steps when it moves. 
Make sure the belts are not loose or too tight.
Make sure the little screws that hold the toothed pully wheels on to the motor shafts are all tight, and that the belts sit on the middle of the pulleys, not pushed up against either edge of the pulley.
Make sure the roller wheels on the rails are not too tight or too loose, they should move nice and smooth without having to force them but tight enough not to wobble about.
Make sure the main frame is properly square and on a flat even surface.

Anel1210, 2019-11-10:
Hi Stuart.

Sorry for the late reply. I had to attend a few other meetings. I am trying
to engrave and cut 3mm mdf wood

StuartB4, 2019-11-10:
Ok. You could probably cut 3mm MDF but it will take a while. When you make the first cut it leaves a small groove in the surface, so after the first 1 or 2 passes the focus point of the laser changes so you will have to refocus. if you had a Z Axis you would just lower the laser down slightly afte a couple of passes.
But without it you have to refocus it by hand, but because the focus rings of a lot of lasers are not very tight and wobble slightly you could move the lens and then it will miss the original burn line. Also the more passes you do the more chance there is of getting brown scorch marks on the surface of the wood around the cut line. That can usually be removed with light sanding around the area.

Anel1210, 2019-11-11:
So in other words. I have to press the home button every time before the
next pass? Because I will I refocus by hand to get it precise?

Anel1210, 2019-11-12:
Hi. Tried lining up manually by hand but it was a disaster

StuartB4, 2019-11-12:
I know what you mean, it's not easy. A Z axis makes it easier. But that means spending more money and adding parts to the machine. It can be done though.

arkypita, 2019-11-12:
Balsa and Plywood are easy to cut also with 2W laser (up to 5mm if well focused) but MDF is a very dense material to cut with diode.

Try search youtube for "laser diode cut MDF", you will find very few videos of people using very powerfull diodes (10-15W).

MDF needs 40-50W CO2 laser to be cut easly.

PS. if you don't wont a motorized Z axis you can also solve with this:
[image]

Anel1210, 2019-11-12:
How does this work and what is it? Do you have any videos on it?

Anel1210, 2019-11-13:
Hi Stuart. Tell me a bit more about how one can upgrade your current
machine? Is it like just attaching a stronger laser on the machine or do
you need to attach a router or what? Can this be done? Like say for example
you do your engraving part then you disconnect the laser and connect a
router or how does that work?

There is not alot of info on this and being new to this, I find it very
difficult to fully understand everything. I learn by trial and error and
with assistance from you guys and guided by you.

Hope you will be able to assist me.

Regards
Anel

[4 more replies]

closed 2020-01-15 by arkypita