arkypita/LaserGRBL #748: Set of images

pcanof, 2019-10-18:
I need to upload a set of images and convert them to gcode. Is there a way to convert together all that images to gcode?

StuartB4, 2019-10-18:
[image]

When you have finished all the images and settings you click Save Program on the File menu.
The only thing is, you end up with 1 big list of GCode with all the offsets included.

You can seperate the blocks of GCode in to the individual image files, just look down the list for where one image ends and the next one starts. As they all start with G90 you just look for that in the code, you can do it in Notepad by opening the GCode file and click on the Edit menu and click Find. Then type in G90 and press Enter. That is the first image, the end of the first one is the last line before the next G90.
Then press Find Next and it will find the start of the code for the second image and then the next and so on.

StuartB4, 2019-10-19:
Just realised, if you load the first then click append to load the others, load them all in at the same place with NO offset. Then they will all load in the same position over the top of each other. But they still give you one big list of GCode when you click Save Program on the file menu. Then you can still split the code in to the seperate images the same way, just look for G90 to find the start of an image and copy and paste everything down to and including the last line before the next G90.

closed 2019-10-25 by pcanof