arkypita/LaserGRBL #749: Photo burning problem

TheTchief, 2019-10-19:
Hello,

I want to print a photo on a funeral urn but the result is not good.

What settings do I need to use to burn grayscale?

Thank you !

StuartB4, 2019-10-19:
Your laser needs to be able to adjust the power up and down (PWM/TTL mode).
If you Start LaserGRBL and click Connect, then go to the GRBL Menu top left, then click Grbl Configuration. You should have a popup like this:

[image]

Check the 3 settings I have highlighted and see if they are the same. If they are not just click on the value box with the numbers in for each one and change the value to the one you see there.
Then click the Write button and Close.

If you don't have them $ settings you have an older version of GRBL. 

Also in Target Image Setting, if it shows M3 for Laser on change it to M4 in the drop down box.

[image]

StuartB4, 2019-10-24:
Did you have any luck with it?

TheTchief, 2019-10-26:
> Répondre à : arkypita/LaserGRBL <reply@reply.github.com>

Hello,

I am starting to understand your software better and to have results.

Given that there are many options, I would like to know where I could find a complete user guide (ideally in French) to make the most of its many possibilities?

Thanks for your help !

Denis Cloutier
Quebec, Canada
denis@top-focus.ca

StuartB4, 2019-10-26:
You could join the LaserGRBL Facebook group for more help, also Youtube videos. I've not seen any complete user guides anywhere either French or any other language.

closed 2020-01-15 by arkypita