arkypita/LaserGRBL #750: problem on a vectorized image

Paoloc66, 2019-10-19:
When I vectorize an image, on the screen it seems perfect, but when I go to print it the outline is slightly shifted down and to the right. I don't know if it's due to some settings to calibrate.
I don't think it's a laser problem as using other programs, printing is perfect.
In some cases it looks like a 3D effect ...
[image]

Paoloc66, 2019-10-21:
[image]

In this image we can clearly see that there is a displacement of the outline

arkypita, 2019-10-21:
Hi @Paoloc66 
This is a know Issue of LaserGRBL that is difficult to resolve and that is much more evident with small images and low resolutions (low number of lines per millimeter).

When we use vector+filling the process will first produce the vectorized path, then the vectorized path is used to produce a B/W bitmap that the line2line algorithm will use (as any raster image) to produce the filling effect.

Since the line2line algorithm starts from 0,0 and goes on by the step settled by lines/mm this output could not match exactly the border drawn with vectorized path.

A better implementation will require to generate filling of vectorized path by producing vector path/patterns for filling (something like this https://wiki.evilmadscientist.com/Creating_filled_regions) but unlucky this is not easy to implement in code.

closed 2019-11-03 by arkypita

IgnasGi, 2020-06-03:
The same problem.  Outline (stroke) is shifted up a little.
The small images (on leather) looks bad.
Have any ideas how to fix that? 
Engrave outline and fill separately is .. too complicated :)
[image]