arkypita/LaserGRBL #751: Buttons Macros

tommeyers, 2019-10-20:
I have added two macro buttons.
1) In low power traces the box that encloses my vectors; All OK :
G0 X[left] Y[bottom] F3000
F2000
M4 S30
G1 X[left] Y[top]
G1 X[right] Y[top]
G1 X[right] Y[bottom]
G1 X[left] y[bottom]
M5
G0 X0 Y0 F3000
2) One that turns on the laser in low power at home for 6 seconds for focusing; Not OK never lights; here it is:
M5            ;off
G0 F500   ;speed
G0 X0 Y0 ;Home
M3 S50      ;on low power
G04 P6.0000 ;pause
M5                ;off

What do I need to do to make this work
Thanks, Tom

StuartB4, 2019-10-20:
If you are using GRBL 1.1 try this. 

$32=0
G0 F500 ;speed
G0 X0 Y0 ;Home
M3 S50 ;on low power
G04 P6.0000 ;pause
$32=1
M5

tommeyers, 2019-10-21:
That did it. 

Thank you, Tom

 

Links:
------
[1]
https://github.com/arkypita/LaserGRBL/issues/751?email_source=notifications&amp;email_token=AD6GJYN3C7NEY5QNSNZDYG3QPTNMVA5CNFSM4JCWHRPKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEBYV5FA#issuecomment-544300692
[2]
https://github.com/

StuartB4, 2019-10-24:
You're welcome.

VirtualGit, 2019-12-16:
Not sure about this but, I would instead use this :
Toggle ON :
```
G1 F1000
M3 S7
```
Toggle OFF :
```
M5
G0
```

because `$32=0` and `$32=1` may cause EEPROM writes (not sure about this) and EEPROM write cycles are limited (maybe arround 100 000 cycles). We may never reach this limit, but if we can avoid increasing the write counter...

closed 2020-01-15 by arkypita