arkypita/LaserGRBL #752: Software question

dhg6, 2019-10-21:
Hello

I can't seem to find a way to show in low power a box to see where to center your item to be etched.  I see a lot of examples of people getting a rectangle, low power laser light trace area to center your work.  Is their a way to do this with LASERGRBL?  I look forward in hearing from you.

Sincerely

David H. Grothe

StuartB4, 2019-10-21:
Right click on an empty part of the main window at the bottom and select Add Custom Button.
[image]

You will get this popup.
[image]

Then Read this.

Function: Trace a box around the image you have prepared so you
know exactly where on the workpiece the laser will go when burning.
Type: Button. - Single press.

M4 is Laser On.
S2 sets the laser on low power. This can be any number between 1 and 255.
1 = lowest power, 255 = highest power.

F500 is the speed the laser will move around the box, you can change this
to a higher or lower number to increase or decrease the speed.
M5 is Laser off when completed.

GCode:

M4 
S2
F500
G0 X[left] Y[bottom] 
G1 X[right]
G1 Y[top]
G1 X[left] 
G1 Y[bottom] 
G0 X0 Y0 F500
M5

Tool Tip: Framing - Laser On Low Power. S2. Speed 500

Enabled: Idle Program.

StuartB4, 2019-10-21:
Another Custom Button.

Function: Find center of prepared image. Click On - Click Off.
Type: TwoStateButton.

GCode box 1:

$32=0
G0 X[left+width/2]
Y[bottom+height/2]
G1 M4 S5 F900

GCode box 2:

$32=1
G0 X0 Y0
M5

Tool Tip: Find Center Of Image. Click On - Click Off.

Enabled: Idle Program.

dhg6, 2019-10-21:
Thank you so much.  I also would like to know, is the s max were you actually control laser power as I have a 15 Watt laser and I want to choose lower power for wood.  Thanks in advance!

StuartB4, 2019-10-21:
Yes, 0 lowest 255 highest.

closed 2020-01-15 by arkypita