arkypita/LaserGRBL #753: Moving the path around the workspace

WattsC-90, 2019-10-21:
I am not sure if it is a feature request, or something I cannot find in the interface, but I have imported an SVG into the software (no editing/import wizard/tools), just plops it straight in. This auto allocates it to 0,0. This cannot be moved or relocated. 

Is it possible to move this item around? Currently I have to turn my laser on/off until I can get the 0,0 approximately where I want it to be, not exactly the precision I was hoping for!

WattsC-90, 2019-10-21:
Also I am not sure if you can define this in the software, but actually how big the "workspace" is! to constrain the software screen rather than having a dynamic screen

StuartB4, 2019-10-21:
> Also I am not sure if you can define this in the software, but actually how big the "workspace" is! to constrain the software screen rather than having a dynamic screen

If you have the GRBL Configuration Settings set for the actual maximum workspace size, when you try to resize the image in Target Image Settings it will put a red warning box around the W or H setting if you go over the number in GRBL. Just so you know it can't be any bigger.
For example, my max work area is 285 mm x 385 mm. 
If I try to resize an image in Target Images Settings bigger by even 1 mm in either W or H boxes, a red box appears around the number.

WattsC-90, 2019-10-21:
If you try with a pure SVG, it doesnt bring up the import image tool, just plops the path into the workspace. And the issue is with using the image import (used it before for something else) is that once its imported, it cant be moved around again. I think it would help if the image was imported, and the workspace was still based on the components you drag in (images/paths/etc), but the workspace got converted/locked when you hit go/play.

The svg I am trying to burn is attached, unfortunately it wont let me attach the plain .svg, sorry.

[Laser_WelliePuller_Label_90.zip](https://github.com/arkypita/LaserGRBL/files/3751428/Laser_WelliePuller_Label_90.zip)

Also, thanks for the settings tip for workspace, hadnt seen that in there!

StuartB4, 2019-10-21:
The only way I can see is to add the offset to the deisgn when you make it.
Make the document larger but keep your design the same size and then move it X amount and Y amount in the document. 

The first one is your original, the second was with the document properties set to 100mm by 100mm and the design moved right and up X 30 and Y 30. Obviously you can put it where you want inside the document. 
[image]

[image]

WattsC-90, 2019-10-21:
The annoying thing is.. that the CAD drawing i exported to SVG doesnt have the text there in relation to 0,0.. obviously the svg is coming out relative to 0. Just all very confusing and frustrating! 

This is why it makes sense to add the tools into the UI to move them about, then lock. Even more so when you try to append a file! How thats meant to work i have no idea, but it ended up overlapping them and I gave up. I would be very keen to know if it is something that could be worked on with minimal effort, or if it was something impossible (or what the case was), because its a bit of a pain when youre trying to locate the burn on the work repeatably.

[image]

StuartB4, 2019-10-21:
I don't think Arkypita the LaserGRBL developer can do much with it because the code he used to support SVG files was already written by someone else, he just added it to the program. As you can see here:
https://github.com/arkypita/LaserGRBL/issues/451
7th comment from the bottom.

StuartB4, 2019-10-22:
I tried Lightburn software just to do SVG files, works fine, lets you move the object around and resize if you need to. They have a 30 day trial version with no restrictions.

WattsC-90, 2019-10-23:
yeh lightburn does the trick, confusing myself with 0,0 now haha given it can move around in lightburn depending on your settings!

preferred the ui in lasergrbl though, much simpler! shame.

Thanks for your help, hopefully Arkypita can pitch in at some stage and comment

StuartB4, 2019-10-23:
I know what you mean about 0,0.
I home my machine and it's all set right and ok. Then I load an image in.  If I tell an image to have an offset of say X 10mm from the left hand edge of the main window, in LaserGRBL it moves the image so that the left hand edge of it is 10mm from the left edge of the window. Which is what I want.
If I set the image position in Lightburn to X 10mm it places the CENTER of the image 10mm to the right of the left hand edge of the main window, so the rest of the image is outside the workspace.  May be me missing something because I'm new to Lightburn.

Yeah, I prefer LaserGRBL but thought I'd try Lightburn. Probably wont use it that much but it is handy for SVG stuff if it does what you want. Good luck with it.

StuartB4, 2019-10-23:
Haha. Just fixed it. Because I use it on a 15" Laptop some of the menus can't be displayed.

This was hiding. Lets you pick the position. Just clicked the bottom left circle and it works the same as LaserGRBL.

[image]

closed 2020-01-27 by arkypita