arkypita/LaserGRBL #754: problem uploading an image

orlando769, 2019-10-25:
[setting.pdf](https://github.com/arkypita/LaserGRBL/files/3770041/setting.pdf)
good evening.
I'm having trouble uploading an image.
the pdf shows the progress and the result.
Thank you

arkypita, 2019-10-25:
You have taken a big image full of detail, than you request to mark that image to a size of 2x3 mm at 6 line/mm. A total of about 12-18 lines.

What did you expect to have as result? A miracle?

arkypita, 2019-10-25:
Moreover seems that your engraver cannot reach that resolution, has you can see the preview show sixteen Lines but the engraver trace only 8 lines.

This means that write your motor cannot step so thin as you request. You are going to push your hardware very over its limits.

Try engraving somerhing simpler.

closed 2020-01-27 by arkypita