arkypita/LaserGRBL #757: test size button

wisaman, 2019-10-27:
Hello friend, thank you very much for your program. I would like to ask you to add the test size button to the program. I offer my respect and greetings.

StuartB4, 2019-10-28:
Hi Wisemanx. What do you mean by test size ?

wisemanx, 2019-10-30:
test area scan

StuartB4, 2019-10-30:
Do you mean to see the area where the laser will go when it burns the image you have loaded.?

wisemanx, 2019-10-30:
yes

wisemanx, 2019-10-30:
[image]

StuartB4, 2019-10-30:
What program are you using, this place is for LaserGRBL.

If you use LaserGRBL this is what you need to do.

You have to make a custom button. 
Right click on an empty part of the main window at the bottom where the buttons are and select Add Custom button.

You should get this pop-up.

[image]

Then do this:
Make sure the top drop down arrow is set to Button.

Copy and paste this code in to the GCode box

M4 
S2
F500
G0 X[left] Y[bottom] 
G1 X[right]
G1 Y[top]
G1 X[left] 
G1 Y[bottom] 
G0 X0 Y0 F500
M5

Add a Tool Tip in the next box, it can be anything you want so you know what the button does
Tool Tip: 
Mine is this  Framing - Laser On Low Power. S2. Speed 500

In the GCode the S is the laser Power(0 to 255), the F is the speed.

Change Enabled from Always to Idle Program with the drop down arrow.

You have to be connected for the button to work.
Load you image as normal, prepare it for burning but before you press Start click the new custom button and it will draw an imaginary box around the image with the laser on low power.
That lets you see where it will be on your workpiece.

wisemanx, 2019-10-30:
my friend thank you very much for your help. I offer my respect and greetings.

wisemanx, 2019-10-30:
my friend, the program does not follow the places where the transaction. and does not change color. cursor at X0, Y0 coordinate

wisemanx, 2019-10-30:
[image]

StuartB4, 2019-10-31:
It's not supposed to change colour,  it should just move around the image lines as it cuts or engraves.
Are you saying the gcode is moving up the screen on the left but the cursor is not moving?
Does your laser move?

arkypita, 2019-10-31:
I confirm: It's not supposed to change colour.
Cursor will be drawn in actual position, and X/Y coord in upper-right will show the position.

Position information is deduced  by communication sent by hardware (grbl running on arduino) to the PC via USB/Serial port. I guess that your hardware is not sending positions or it sends in a wrong/out of specific format.

Can you copy-paste a final piece of your session log?
Menu ? -> Open Session Log

wisemanx, 2019-11-03:
The laser is moving but the cursor is not moving.

wisemanx, 2019-11-03:
03.11.2019 21:04:11	OpenFile    	Open D:\DESEN\Artherapie_ca.jpg
03.11.2019 21:04:27	SetStatus   	Machine status [Run]
03.11.2019 21:04:34	SetStatus   	Machine status [Idle]
03.11.2019 21:04:35	EnqueueProgram	Running program, 22028 lines
03.11.2019 21:04:35	SetStatus   	Machine status [Run]
03.11.2019 21:33:58	ProgramEnd  	Job Executed: 22028 lines, 0 errors, 29 min
03.11.2019 21:34:03	SetStatus   	Machine status [Idle]

wisemanx, 2019-11-03:
arduino grbl is moving in version 1.1. but grbl 0.9j does not move.

arkypita, 2019-11-05:
Nothing useful from session log.

I'm pretty sure this is some variability of the status message that grbl sends to the PC that I didn't expect to meet, which doesn't make it possible for me to interpret the message.

If you want to help me debugging and solving this issue please tell me the value of $10 config variable from Menu "File" - "Grbl Configuration" (when connected and Idle) in version 0.9j

Also you can record the whole communication following this instruction:

> LaserGRBL contains a diagnostic mode can be set by command prompt that log the whole communication process.
> 
> If you start LaserGRBL from command prompt or by a shortcut with "comlog" argument "C:\Program Files (x86)\LaserGRBL\LaserGRBL.exe comlog" the program will record a file called "comlog.txt" located in "C:\Users\YourName\AppData\Roaming\LaserGRBL".
> 
> In comlog file you can find all the serial data transmitted/received from LaserGRBL.
> Here is how a valid comlog look like: https://github.com/arkypita/LaserGRBL/files/2636346/comlog.txt

arkypita, 2019-11-05:
Is it possible that your $10 status report is configured to don't report machine position? (i.e. zero)
If so, please set it to 3.

https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#10---status-report-maskbinary

sinfocomp, 2019-11-27:
[comlog.txt](https://github.com/arkypita/LaserGRBL/files/3897459/comlog.txt)
[sessionlog.txt](https://github.com/arkypita/LaserGRBL/files/3897460/sessionlog.txt)
Hello, attached are the files, hope this helps connect lasergrbl to my Mega cleanly
thanks

quotelawrence, 2020-01-04:
I have the same software and it shows a feature to test the area before laser activation very low beam so you can see the light and watch to make sure your project is located properly on the material I am not sure if that button was created but if you were able to program a feature that would take the boundary and run a square pattern around that area did it work ?

StuartB4, 2020-01-04:
Start LaserGRBL and click on Connect, then follow this:

Right click on an empty part of the main window at the bottom and click Add Custom Button.
[image]

[image]

This button will draw  a box around the image you have prepared with the laser on low power, so you
know exactly where the laser will go when burning.
Type: Button. - Single press.
Add the code below to the GCode box. You can copy and paste it from here:

M4 
S2
F500
G0 X[left] Y[bottom] 
G1 X[right]
G1 Y[top]
G1 X[left] 
G1 Y[bottom] 
G0 X0 Y0 F500
M5

Tool Tip: Framing - Laser On Low Power. S2. Speed 500

Enabled: Idle Program.

M4 is Laser On.
S2 sets the laser on low power. This can be any number between 1 and 255.
1 = lowest power, 255 = highest power.

F500 is the speed the laser will move around the box, you can change this
to a higher or lower number to increase or decrease the speed.
M5 is Laser off when completed.

-----------------------------------------------------------------------

DonLeon13, 2020-01-10:
I used the code you showed, but my laser dont turn on it drives around 

> Start LaserGRBL and click on Connect, then follow this:
>
> Right click on an empty part of the main window at the bottom and click Add Custom Button.
> [image]
>
> [image]
>
> This button will draw a box [...]

I did everything exactly as you described it, but my laser does not turn on.
He does the lap and shows me the size of the engraving but without a laser.
What could that be?

StuartB4, 2020-01-10:
Change the M4 in the GCode box to M3 click save and try it again.

DonLeon13, 2020-01-11:
I try it with M3 but it doesn't work

StuartB4, 2020-01-11:
Which version of GRBL do you have?

[image]

DonLeon13, 2020-01-11:
1.1f

StuartB4, 2020-01-11:
Do you have these 3 settings set like this.
$30 can be 255 instead of 1000 and it still should work.

[image]

DonLeon13, 2020-01-11:
OK I try 👍

DonLeon13, 2020-01-11:
It moves but the Laser don't turn on

DonLeon13, 2020-01-11:
Now it works I have set the laserpower to s10
Thanks for your help 👍

StuartB4, 2020-01-11:
Ok, thats good. :+1:

closed 2020-01-27 by arkypita