arkypita/LaserGRBL #761: SVG size

GusanoGris, 2019-11-05:
Now, to be able to specify the size of the SVG image, you have to edit the file and change the width and length in the header:
[image]

I think that being able to define these parameters in the file load would be important

[image]

appieollo, 2019-12-10:
YES, that would be one of my first wishes !!

GusanoGris, 2020-03-16:
Sorry, I manage 0 with GitHub, so I send a zip with a solution to the problem of the size of svg files based on a percentage of the width and length of the image.

It is working for me so, if you like it, it could be incorporated.

I suppose the code could be improved and adjusted more to the standard, but that is up to you.

Greetings.
--------------------------------------------
Lo siento, me manejo 0 con GitHub, así que envio un zip con una solución al problema del tamaño de los ficheros svg basada un porcentaje sobre el ancho y largo de la imagen.

A mi me está funcionando así que, si te parece bien, se podría incorporar.

Supongo que el código podría mejorarse y ajustarse más al estándar, pero eso ya a vuestra elección.

Un saludo.
---------------------------------------
[SvgConverter.zip](https://github.com/arkypita/LaserGRBL/files/4337709/SvgConverter.zip)

arkypita, 2020-03-16:
@GusanoGris thanks for your contribution
I'll give a look as soon I have time. Best regards

brybalicious, 2025-08-04:
@arkypita have you managed to incorporate this?