arkypita/LaserGRBL #763: Found Mailware in Download

Pix4hawk, 2019-11-05:
my antivirus found Mailware  HEUR/APC !??
[image]

StuartB4, 2019-11-05:
It's a false positive, unless you downloaded it from somewhere other than LaserGRBL website.
It has something to do with certificate signing I think.

Pix4hawk, 2019-11-05:
Thanks for quick answer!

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/arkypita/LaserGRBL/issues/763?email_source=notifications\u0026email_token=AEKX7XRX24GEK3MZCS2H4MTQSHOS3A5CNFSM4JJINYC2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEDEKWWQ#issuecomment-550021978",
"url": "https://github.com/arkypita/LaserGRBL/issues/763?email_source=notifications\u0026email_token=AEKX7XRX24GEK3MZCS2H4MTQSHOS3A5CNFSM4JJINYC2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEDEKWWQ#issuecomment-550021978",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

StuartB4, 2019-11-06:
Here's a better explanation from the author of LaserGRBL.
https://github.com/arkypita/LaserGRBL/issues/70

closed 2020-01-27 by arkypita