arkypita/LaserGRBL #764: Small output size improvement

arkypita, 2019-11-06:
Hi friends!
I want to open a discussion about raster import when producing small image as output.

At the state of the art if in LaserGRBL you open an hi-res full of detail, and you want to engrave it to a very small output surface (for example 5x3mm) LaserGRBL will scale down the original raster image to the size of the output, losing a lot of detail and producing an useless output in vectorization fase.

[image]

[image]

[image]

This makes LaserGRBL able to work well only on medium-large output sizes (starting from 30x30mm upwards) or in any case with source images whose resolution/size/details is comparable with the output dimensions.

arkypita, 2019-11-06:
With few changes I can modify LaserGRBL to avoid scaling down the raster image, working on the native image during vectorization and scaling down only the final vectorized image.

This should make it possible to obtain very detailed images (provided that the laser is sufficiently precise) even on small output sizes.

[image]

What do you think about it?

StuartB4, 2019-11-06:
Sounds like a great improvement.

arkypita, 2019-11-14:
I have done some test with this, but I am not satisfied with the results.

Maintaining the resolution/quality of the original image for vectorization and then scaling the result, easily leads to "too detailed images" that the laser, however precise, marks as a "black spot".

In the example above (a galeon to be engraved 4mm x 3mm) it sounds obvious that it will be impossible to get a decent result.

Perhaps there could still be an improvement trying to maintain a resolution a little higher than the current one, but it is quite obvious that going to resize downwards a loss of detail is necessary to maintain the image engravable.

StuartB4, 2019-11-14:
Oh ok. That's a shame.

arkypita, 2019-11-14:
These are the results of my test

Source image:

[image]

Using target size with different width: 3mm, 5mm, 8mm, 12mm, 20mm, 30mm

Bottom line is vectorization generated by the actual LaserGRBL algorithm (resize down the raster image, smoothing and loosing information, and then vectorizing).

Top line is generated with a new algorithm that vectorize the source image at the full size, and then scale down the vector path without losing quality.

[image]

Is it possible to appreciate the differences looking at the smallest version of the image (3, 5, 8, 12mm)

[image]

The old algorithm lose details while the new one preserve them.

However, the difference on the results is not so appreciable. A better quality can be noted only on 12mm image (the 4th from left). In smaller ones the quality of stepper, the width of the laser spot etc waste all the details and quality of the image.

[image]

arkypita, 2019-11-14:
If anyone want to test with its own engraver, here is the gcode.

[compare.zip](https://github.com/arkypita/LaserGRBL/files/3848878/compare.zip)

StuartB4, 2019-11-15:
That's a lot better. I don't have anything to try it on at the minute but it looks really good.