arkypita/LaserGRBL #765: Can't get PWM connected on new laser module

chrislarkins, 2019-11-06:
I recently bought a cheap laser engraver from Banggood with a 3000mw laser, I want it for cutting balsa and ply but is just wasn't powerful enough. I purchased a 15W laser from eBay but I'm having trouble connecting it to my control board, my board used 2 wires to control the old laser and this worked well and switched the laser on/off as required and fired up in test mode.
I can't get my new laser working correctly, it seems to be always on unless I hit the reduced power button, I have tried various combinations and the nearest I have come to getting it working is in reverse i.e the laser is on when stationary but switches off when it should be cutting.
Can anyone please help?
[image]
[image]

StuartB4, 2019-11-06:
The single red with the plug on that goes to the 3 pin socket on the laser driver board needs to get it's signal from Pin D11 on the Arduino Nano on the main control board.

 I see the main board has a 3 connector (is it P5) but has no socket soldered in to it. If you have a multimeter you could check using the continuity setting to see if pin D11 on the Nano board is connected to 1 of the 3 positions.

StuartB4, 2019-11-06:
Or you could upgrade you main controller board to something like this, it has connectors for 2 wire and 3 wire (TTL/PWM)
https://www.tomtop.com/p-e8033.html#flow_review

chrislarkins, 2019-11-06:
Hi, I measured the voltage coming from D11, with the laser power set to 100% on lightburn the voltage is about 7.5v, with the power set to 10% the voltage is about 11.5v . Does this sound about right?

StuartB4, 2019-11-06:
I just checked mine using lightburn and while it's burning an image in greyscale mode the voltage goes from 0 to 5v.
0 being laser off and 5V laser on 100% power.

StuartB4, 2019-11-06:
You might find this bit of info helpful.
https://www.arduino.cc/en/tutorial/PWM

arkypita, 2019-11-07:
> Hi, I measured the voltage coming from D11, with the laser power set to 100% on lightburn the voltage is about 7.5v, with the power set to 10% the voltage is about 11.5v . Does this sound about right?

As arduino is a 5V device none of its pin can exceed 5V. If you measure 7 to 11 volt on pin D11 two are the options:

1) your multimeter is broken, or you are not using it properly (please note that measuring PWM is not so simple as measure a battery)

2) your pin D11 is connected to something that force it to be 7 to 12v. This is not correct and should be avoided as D11 is an output pin. Are you sure you have plugged it to the right connector? Try test it with PWM cable detached.

chrislarkins, 2019-11-08:
Many thanks for your help, I bought a new controller which was only £17.89 with free postage on Amazon, it turned up today and after swapping it over I have success. I did have to add a ground as described above but after that it worked straight away. I did a test engraving on 3mm birch ply and I'm really happy with the result, I also did a cut and it took 3 passes to get all the way through, with my old laser it took over 25 passes in 3mm birch ply.
https://www.amazon.co.uk/gp/product/B07RQZHM44/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

chrislarkins, 2019-11-08:
https://youtu.be/wPTy8IuY1FY

StuartB4, 2019-11-08:
Hi Chris. 
That's good to know it's all working.  :)
Thanks for letting us know.

closed 2020-12-16 by arkypita