arkypita/LaserGRBL #766: can't get the machine to work

baselbaransi, 2019-11-06:
hello all, 
I am new to this and i need your help.
i bought the laser machine with the board "Makerbase MKS DLC V2.0)
I connected it to the laser machine and installed the required program to get it work.

[image]
 as you can see in the picture. it is connected, BUT when i click on the arrows the laser does not physically move.

can any one help me with this problem....

thanks a lot.

StuartB4, 2019-11-06:
With LaserGRBL running and your machine connected goto the Grbl menu and click Grbl Configuration. 
You should get a window with all the GBL settings. Like this.

[image]

Can you do a screen shot and upload it here for me to check.
Thanks.

baselbaransi, 2019-11-07:
here is the configuration screen, when i open it i get this error below..

[image]

[image]

the error message.

[image]
the details:
"See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.ArgumentOutOfRangeException: Value of '0' is not valid for 'Value'. 'Value' should be between 'Minimum' and 'Maximum'.
Parameter name: Value
   at System.Windows.Forms.TrackBar.set_Value(Int32 value)
   at LaserGRBL.JogForm.UpdateFMax_Tick(Object sender, EventArgs e)
   at System.Windows.Forms.Timer.OnTick(EventArgs e)
   at System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.9148 (WinRelRS6.050727-9100)
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v2.0.50727/mscorlib.dll
----------------------------------------
LaserGRBL
    Assembly Version: 3.0.18.0
    Win32 Version: 3.0.18.0
    CodeBase: file:///C:/Program%20Files%20(x86)/LaserGRBL/LaserGRBL.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.9136 (WinRelRS6.050727-9100)
    CodeBase: file:///C:/WINDOWS/assembly/GAC_MSIL/System.Windows.Forms/2.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.9147 (WinRelRS6.050727-9100)
    CodeBase: file:///C:/WINDOWS/assembly/GAC_MSIL/System/2.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.9136 (WinRelRS6.050727-9100)
    CodeBase: file:///C:/WINDOWS/assembly/GAC_MSIL/System.Drawing/2.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Core
    Assembly Version: 3.5.0.0
    Win32 Version: 3.5.30729.9135 built by: WinRelRS6
    CodeBase: file:///C:/WINDOWS/assembly/GAC_MSIL/System.Core/3.5.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.9136 (WinRelRS6.050727-9100)
    CodeBase: file:///C:/WINDOWS/assembly/GAC_MSIL/System.Xml/2.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Configuration
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.9136 (WinRelRS6.050727-9100)
    CodeBase: file:///C:/WINDOWS/assembly/GAC_MSIL/System.Configuration/2.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box."

***********************
this pics when there is no error popup.
[image]
[image]

thanks a lot

StuartB4, 2019-11-07:
Have you tried uninstalling it and reinstalling it. 

Can you try this. 
Download this GRBL Settings file from my pcloud account and put it on your desktop.
https://my.pcloud.com/publink/show?code=XZgQmQkZgDS12tMRguFGkr0ufzgqu0PGGKWX

Start LaserGRBL and click connect.
Go to the GRbl menu again and select the GRBL Configuration.
When the window pops up with all the settings click on the Import button at the bottom and browse to the desktop, select the GRBLsettings file on your desktop.

Then let me know what happens.

arkypita, 2019-11-07:
Your configuration seem to be totally weird. A valid configuration look like this:
[image]

arkypita, 2019-11-07:
Follow Stuart hint and try rewrite your configuration with the file he provide.

arkypita, 2019-11-07:
The exception is because when you open config form the grbl configuration is read from the hardware, and the jog form try to self-configure to limits the max value of the speed slider to the max speed in the just read configuration.

It does not protect itself from a misconfigured hardware, so it fail when it read strange values.

baselbaransi, 2019-11-07:
the laser worked, thanks for that i changed the configuration setting an the laser moved.

but a new problem happened, the laser now doesn't carve on the wood or any thing..

i tried the solution of Stuart, is there a way to change the laser power, or maybe i should change the wood type..

arkypita, 2019-11-07:
What is your laser power in Watt?

Stuart configuration use "$30=1000 (Maximum spindle speed)" so if you want max power you should write 1000 in raster import (second window).

baselbaransi, 2019-11-07:
how is $30 related to the power of the laser?

StuartB4, 2019-11-07:
It's the maximum speed of a spindle motor if the machine had one for CNC machining.
When you change it for a laser it works as the maximum power of the laser.

StuartB4, 2019-11-07:
You may also have to tweak the settings in Grbl configuration a bit to match you machines steps/mm for the motors. $100 and $101.
If you tell your machine to jog a certain amount, say 25 mm's, then measure exactly how much it actually moved, if it's wrong then you change $100 for the X axis and do the same for the Y axis with $101.
Open LaserGRBL and make sure you are connected. Open the configuration settings on the GRBL menu as before. Click in the Value box of $100 and change the setting, do the same for $101 and then click the Write button to save the settings in GRBL.

dogs4kids, 2019-12-23:
The software recognizes my  machine and will move the X and Y axis, but it will not turn on the laser. What do I need to do to get this working?
[LaserGRBL_config.txt](https://github.com/arkypita/LaserGRBL/files/3993518/LaserGRBL_config.txt)

closed 2020-01-27 by arkypita

mommameg970, 2020-03-14:
i am having the same problem, but i am not able to get past the dialog box with the error... i have tried this on 2 of my other computers and now cant open the software...