arkypita/LaserGRBL #767: Questions about 500mw blue laser

esdrasramos1988, 2019-11-07:
Hi there folks, 

I'm Esdras Ramos from Brasil, a few weeks ago i decided to mount my own laser engraver/cutter so i could have some fun with it and also make some spare money from whatever it can provide me..

I didnt know much about the power of each kind of laser and i guess i took a very wrong decision into buying a 500mw laser module..  

Allthoughts, i plan to switch it into a 20W laser ( much stronger eh) sooner or later...

if you've read it till here, prolly means your interested in helping me out about the questions i've got while i'm mounting such equipments.

so... there goes my not long questions 
hope to not make any of you mad, since as i said before i'm pretty new on this adventure, 
also counting with you guys to guide me into such an amazing tech ...

1- what can i cut with a 500mw blue laser?
2- what can i draw with it?
3-is there any special program to use on buring photography? (free tho), i tried laser grbl for that and didnt work well
4-in case of a thick wooden plate (3mm i know its allready pretty hard to cut) how many times do I have to pass the Gcode, over and over again to cut off the part?
5- is there a "$" comand to "loop" or some macro button we can place to make it work?
6- i'm having problems on generation of G-code for stuff like drawning anything that is a cartoon and comes from the internet, it doesnt burn properly, like, burns the entire thing very wrong..
7- is it ok to connect a 20W laser on a regular cnc shield? i fear shortcircuits

thanks for your reading, 
once again, dont be mad or anything like that, i'm new on it, 
Esdras Ramos

arkypita, 2019-11-07:
> 1- what can i cut with a 500mw blue laser?

Paper, cardboard max 1mm

> 2- what can i draw with it?

Leather, cardboard, plywood

> 3-is there any special program to use on buring photography? (free tho), i tried laser grbl for that and didnt work well

This is the page of lasergrbl, so i can suggest to use lasergrbl

> 4-in case of a thick wooden plate (3mm i know its allready pretty hard to cut) how many times do I have to pass the Gcode, over and over again to cut off the part?

You cannot cut wood wit 500mW laser

> 5- is there a "$" comand to "loop" or some macro button we can place to make it work?

LaserGRBL can repeat a job multiple times to do more passes.

Gcode have some special syntax to loop/cycle but grbl does not support that codes: https://github.com/grbl/grbl/issues/1154

> 6- i'm having problems on generation of G-code for stuff like drawning anything that is a cartoon and comes from the internet, it doesnt burn properly, like, burns the entire thing very wrong..

We are not magician. Post some pictures and explain your procedure.

> 7- is it ok to connect a 20W laser on a regular cnc shield? i fear shortcircuits

Yes, but you need a right laser driver and power supply.

esdrasramos1988, 2019-11-07:
thanks alot for the explanations sir, 
i did apreciate it, 

-about the question number 6 
i did fix it, went to find out a few config and understood what was happening wrong
now its all ok i guess, as the Screenshot i'm sending...
[image]

about the question 7 
this is the laser i plan to buy... i dont know if its forbiden to post external links, but this is the one i plan to get..

https://pt.aliexpress.com/item/4000214972073.html?spm=a2g0o.productlist.0.0.45b22109xTsAOK&s=p&algo_pvid=9aeb79ae-e864-4d44-ab7e-a750d7a4722a&algo_expid=9aeb79ae-e864-4d44-ab7e-a750d7a4722a-0&btsid=6dbad84d-bcff-40c7-823c-5edec271eb2b&ws_ab_test=searchweb0_0,searchweb201602_4,searchweb201603_55

Once again, 
thanks for all the informations mr, 
hope to learn alot to help others too

Kind Regards, 
Esdras Ramos

arkypita, 2019-11-08:
For your information there are no 20W lasers so cheap.  For my knowledge there are no laser diodes visible above 7W. The most powerfull is NUBM44-V2 and it cost 80$ only the diode.

As you can see [at this link](https://pt.aliexpress.com/item/33059275128.html?spm=a2g0o.detail.1000014.3.5b67dcb0vyWLaA&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.13338.128125.0&scm_id=1007.13338.128125.0&scm-url=1007.13338.128125.0&pvid=87837394-cc24-401b-82ee-892d86ec4592), the same product (20W laser) in the description states "5.5W optical power".

However it seems to be a good upgrade for your engraver. I think that with this laser you can occasionally cut plywood up to 4-5mm.

esdrasramos1988, 2019-11-18:
> For your information there are no 20W lasers so cheap. For my knowledge there are no visible light laser diodes above 7W. The most powerfull is NUBM44-V2 and it cost 80$ only the bare diode.
> 
> As you can see [at this link](https://pt.aliexpress.com/item/33059275128.html?spm=a2g0o.detail.1000014.3.5b67dcb0vyWLaA&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.13338.128125.0&scm_id=1007.13338.128125.0&scm-url=1007.13338.128125.0&pvid=87837394-cc24-401b-82ee-892d86ec4592), the same product (20W laser) in the description states "5.5W optical power".
> 
> However it seems to be a good upgrade for your engraver. I think that with this laser you can occasionally cut plywood up to 4-5mm.

Updating...

Hey mr Arkypita, 
i've bought that "laser 20w" to test it, 
there comes another doubt that came to me..
there's an extra 4th wire named "temp" wire, 
guess it goes for temperarute of the laser diode, like some alarm set..
is there a connection for it on cnc shield?
i've seen it has an input on laser grbl for temperature, but dunno where to connect on the board

p.s.   as my machine's project goes on, 
i'm placing pictures of it soon

thanks for your huge support once again, 
Esdras Ramos

arkypita, 2019-11-18:
Hi Ramos

> i've seen it has an input on laser grbl for temperature, but dunno where to connect on the board

LaserGRBL does not have any input  for temperature. What do you mean?

I think that a "temperature limit safety lock" should not responsability of the PC software, but it should be implemented in hardware/firmware (grbl). This is the only way to have it working with any kind of pc software (LaserGRBL or any others)

> there comes another doubt that came to me..
> there's an extra 4th wire named "temp" wire,
> guess it goes for temperarute of the laser diode, like some alarm set..
> is there a connection for it on cnc shield?

I have seen this wire in the product you link but I cannot understand what kind of signal it provide by reading seller description. Is it a 0-5V on/off signal for temperature over 60°? Is it a linear voltage that can give you a read of the temperature? I don't know. 

You have to understand it by yourself and then decide if you want to use it in a sort of safety circuit.
I think that the CNC controller/shield does not have a dedicated temperature input, so you should design your own. Something was discussed in this topics.

https://github.com/arkypita/LaserGRBL/issues/673

closed 2020-01-27 by arkypita