arkypita/LaserGRBL #768: Help with CNC 3018 Pro

jonvarsol, 2019-11-07:
Hello! I don{t know how to use CNC 3018 Pro, I tried with PC software and cannot engrave with laser. 
It goes only form left ot right and up and down but the below plate doesnt moves, so all I got is a line over and over... It supposed to be the name: Paul Smith and looks in the other picture what I got :( 

[image]
[image]

StuartB4, 2019-11-07:
Are you sure all your electrical connections are ok?
Does the Y axis motor turn at all, can you see the lead screw rotating?
Check the Y axis motor connection. 
Then when the machine is connected go to the Grbl menu in LaserGRBL and select Grbl Configuration.
Look down the list of settings and check the $100 and $101 settings.
They should be about the same.

StuartB4, 2019-11-07:
This may sound daft but is it a new machine? if so, are there any bits of packing left underneath the bed that might be stopping the motor from moving.

jonvarsol, 2019-11-13:
Thanks for your help! and I am so sorry for my late response, I was out of my house. 

- yes, Y axis is working, I can see the lead screw rotating, actually same as X and Z axis.
- it´s a new machine. I already checked everything and there´s nothing underneath the bed :(

Sorry to botter you, but it will be great if you can text me by whatsapp, so I will be ablle to send you some videos... +506 8707-5923

Thanks,

Jonathan
[image]

StuartB4, 2019-11-13:
I don't understand, if the Y axis motor and the lead screw is moving then the bed should be moving.
Did you fix the workpiece, to the bed, there is a chance that it moved when the bed moves.
Also I would change $32 to 1 to enable Laser mode.

jonvarsol, 2019-11-14:
Yes, its really weird! well, I mean, I can see the lead screw rotating when I move with offline control or with the grbl software, but no when is working with laser.

I dont know what you mean with fix the workpiece :(

I changed $32 to 1 and nothing happened.

Here´s another screenshot of my last test (same history).

[image]

jonvarsol, 2019-11-14:
by the way, thanks for your help!

StuartB4, 2019-11-14:
What I mean is, did you fix the item you were engraving to the bed so it could'nt move about with the vibration of the machine.

Changing $32 to 1 gives you more control over the laser power thats all.

You could try changing the steps/mm in grbl configuration settings.
Try changing the Y axis $101 to 1600.

The model number of the machine (3018) refers to the actual engraving area.
X  = 300mm  Y = 180mm. 
The maximum travel for the X axis on yours is set to 200 ($130) from what I can see in the image you posted. So you should be able to set $130 (X) to a bit less than 300.
Also check $131 for the maximum travel of the Y axis and set $131 (Y) a bit less than 180.

arkypita, 2019-11-14:
> Thanks for your help! and I am so sorry for my late response, I was out of my house.
> 
> * yes, Y axis is working, I can see the lead screw rotating, actually same as X and Z axis.
> * it´s a new machine. I already checked everything and there´s nothing underneath the bed :(
> 

Just to be sure: are you telling that if you control the movement with jog all it's fine, and if you send a program the Y does not move (or it move bad)?

Please do this test: set jog step to 20 (2cm), speed ~1000

[image]

Then move the X axis left and right. Does it moves? It moves by 2cm exactly?
Then move the Y axis far and near. Does it moves? It moves by 2cm exactly?

> I dont know what you mean with fix the workpiece :(

Attach to the bed with a bolt/nuts or at least tape.

StuartB4, 2019-11-14:
Forgot to mention $20. 
If you set the Max Travel on $130 and $131 also make sure $20 is set to 1.
It enables Soft Limits so the Laser or the Bed don't go too far.
If you do try to tell it to move more than it's set to it will just not move and will say "Travel exceeded" in the window on the left where the GCode scrolls up.

jonvarsol, 2019-11-15:
StuartB4 and arkypita, thank you so much for your help.

I found the solution... Y and Z cables were changed. You may be laughing ;), It´s my firts machine and I had not experiense. I really appreciate your time.

StuartB4, 2019-11-15:
I'm not gonna laugh...........much. : )
We all have to start somewhere, glad you sorted it, and glad to help if I can.
Good luck with it.

closed 2020-01-27 by arkypita

locovaldes, 2020-11-03:
Hi i have a 3018 Pro , and the laser was working good  and now sometime burn and sometime nothing, or do like small draw no normal
[image]

locovaldes, 2020-11-03:
[image]
[image]

locovaldes, 2020-11-03:
first was and now i try to do the 800 speed test and see what i got