arkypita/LaserGRBL #769: Connection Problem

Kapelaan, 2019-11-07:
I just got my new 3018 Pro Max laser engraver. I have installed LaserGRBL v3.0.18 on my computer, but the program don’t want to connect to my cutter. Can you please help?

StuartB4, 2019-11-07:
Have you tried other programs with any success?

Did it come with a disk with USB drivers on it.

Is it showing up in the Windows Device manager:
You may need to install the CH340 driver.

[image]

Kapelaan, 2019-11-07:
I have also tried engraver-Master and Laser-Engraver, but with no success.

I got some of the driver with the machine on a USB-stick.

I have installed the CH340 driver. The Device Status at the Windows Device Manager is saying: "The device is working properly", but still no program wants to connect to the maschine

StuartB4, 2019-11-07:
Have you tried different USB ports, or a different computer?
Is the Baud rate set to 115200.

Kapelaan, 2019-11-08:
I have tried different USB ports as well as a different computer.

Yes, the Baud rate is set to 115200

StuartB4, 2019-11-08:
Does the control main board have a makers name on it?

StuartB4, 2019-11-08:
Does it have a hand controller so you can use it without connection to a computer.
I do know that usually you can't use it if both the hand controller and the pc are connected at the same time.

Kapelaan, 2019-11-11:
Thank you so much!!! That's it! The hand controller was still connected to the machine. The moment that I have removed it, my program was connecting to the machine. Thank you!

closed 2019-11-11 by Kapelaan

StuartB4, 2019-11-11:
Oh right, that's good to know. Glad you got it sorted.