arkypita/LaserGRBL #773: Fitting drawing on wood shape

baselbaransi, 2019-11-14:
Hello all,
How can i know where the laser will burn the wood exactly?
It there a way to see it on the screen on the computer.
For example, lets say i want to burn a text on a round piece of wood, and want the text to be exactly in the middle, how can i do that?
Or on a rectangular shape wood, i want to see where i will be burning the text/photo according to the size of the wood?..

Is there a custom button to do that? 

Thanks in advance.

arkypita, 2019-11-14:
Look simple to do with a custom button.
https://github.com/arkypita/LaserGRBL/issues/306

baselbaransi, 2019-11-14:
How can i add a custom button?

arkypita, 2019-11-14:
A google search can find what you need
www.google.it/search?q=lasergrbl+add+custom+buttons

baselbaransi, 2019-11-14:
How hard is it to write a Gcode?

arkypita, 2019-11-14:
I have already linked you a page with dozen of custom button samples, including what you asking for: https://github.com/arkypita/LaserGRBL/issues/306

baselbaransi, 2019-11-14:
Thanks a lot, i will take a look at it

StuartB4, 2019-11-14:
Start LaserGRBL and click on the Connect button.
Go to the bottom of the main window and RIGHT click on an empty part of the bar where the other buttons are.
Select Add Custom Button. You can copy and paste the GCode in this text for each button you make.

Function: Trace a box around the image you have prepared so you
know exactly where on the workpiece the laser will go when burning.
Type: Button. - Single press.
Make sure you are connected and your image is prepared for burning, then use the button.

M4 is Laser On.
S2 sets the laser on low power. This can be any number between 1 and 255.
1 = lowest power, 255 = highest power.

F500 is the speed the laser will move around the box, you can change this
to a higher or lower number to increase or decrease the speed.
M5 is Laser off when completed.

GCode:

M4 
S2
F500
G0 X[left] Y[bottom] 
G1 X[right]
G1 Y[top]
G1 X[left] 
G1 Y[bottom] 
G0 X0 Y0 F500
M5

Tool Tip: Framing - Laser On Low Power. S2. Speed 500

Enabled: Idle Program.

-----------------------------------------------------------------------

Function: Switch Laser on low power for focussing.
Type: TwoStateButton.
Click On - Click Off.
Make sure you are connected then use the button.
This button is only valid if you have GRBL 1.1.

$32=0 switches off laser mode. $32=1 switches laser mode back on.
M4 = Laser on. M5 = Laser off. S = Laser power setting.

GCode box 1:

$32=0
M4
S3

GCode box 2:

$32=1
M5

Tool Tip: Laser ON (Laser Mode=OFF) / Laser OFF (Laser Mode=ON)

Enabled: Idle.

-----------------------------------------------------------------------

Function: Find center of prepared image.
Click On - Click Off.
Type: TwoStateButton.
Make sure you are connected and your image is prepared for burning, then use the button.
This button is only valid if you have GRBL 1.1.

GCode box 1:

$32=0
G0 X[left+width/2]
Y[bottom+height/2]
G1 M4 S5 F900

GCode box 2:

$32=1
G0 X0 Y0
M5

Tool Tip: Find Center Of Image. Click On - Click Off.

Enabled: Idle Program.

-----------------------------------------------------------------------

Function: Flash Laser.
Type: PushButton. 
Press and release to flash laser on and off. 
Make sure you are connected and then use the button.
This button is only valid if you have GRBL 1.1.

GCode box 1:

$32=0
M4 S3

GCode box 2:

$32=1
M5

Tool Tip: Press and Release To Flash Laser.

Enabled: Idle.

brakthehun, 2019-11-19:
Laser GRBL also has overrides. If you click the numbers in the lower right,, you can turn the laser power way down so it does not burn and turn the linear speed up so it runs through the program fast.
If you're happy with it,, then you can reset the overrides by clicking on the numbers in the lower right again and clicking the reset button on each changed command.

closed 2020-01-27 by arkypita