arkypita/LaserGRBL #774: Request: Pause before cuting

FrancoisL2007, 2019-11-16:
Hi,
I have a 5w Laser. when i cut white paper, the laser is reflected and do not cut the paper for a few millimeters. it would be nice to allow the laser to wait a short time (maybe 0,05s) to start cutting then make the laser move... 
thank you,
Francois

sinfocomp, 2019-11-16:
Edit your post procesoor

FrancoisL2007, 2019-11-16:
Thank you but I don’t know how to do that. Is it in laserGRBL? Which command?

Grouic, 2019-12-09:
Try to add a "G4 P0.05" after each laser activation in your GCODE

FrancoisL2007, 2019-12-09:
Thank you for your reply.
I have to create a gcode with LaserGRBL then edit the gcode with a text editor then reload the gcode in LaserGRBL.
Is that ok or is there a way to make it automatically?
François

Grouic, 2019-12-09:
It's OK, I haven't found the automatic way for now.

FrancoisL2007, 2019-12-09:
If you find it, tell me!
Thank you so much!
François

Grouic, 2019-12-10:
I only found it on JTech plugin for inkscape (and it only work with M3 order) 
M05 S0

G90
G21
G1 F1000
G1  X102.9135 Y75.7017
G4 P0 
M03 S1000
G4 P0.05**000000074505806**
G1 F750.000000
G2 X102.1077 Y75.7816 I0. J4.1023
G2 X101.6961 Y75.9455 I0.2412 J1.204

The duration is a little bit strange

arkypita, 2019-12-10:
This feature has been requested several times, and I have already included it in the development roadmap: https://github.com/arkypita/LaserGRBL/issues/64

It is not difficult for me to add it, the fact is that those who ask it never take into consideration some details that I consider important.

First there is an issue/conflict with the automatic cooling cycles. When cooling cycles are activated LaserGRBL send suspend/resume commands at periodic times. It is not possible to insert the same power-on delay when resume.

Second there are two/three modes to turn on-off laser: M3-M4 and M5, but also using S0-S255 (power to zero/max), and also perform G0 for rapid move instead of G1/G2/G3.

So, for example, a program can send an M3 command only at the beginning of the program, and then work only with changes of S-power or with G0/G1 to control laser on/off.

I am not sure of that, but in gcode generated by LaserGRBL i sometime use such kind of optimization, that make it more complex that simply add G4 after M4.