arkypita/LaserGRBL #778: not burning

WHAAWHOO, 2019-11-21:
Was working on project and noticed that lines were getting weak, watched for a little bit and then laser was on but not burning.  Quit project then tried another simple project axis was working correctly laser shutting on and off but again nothing burning.  Switched out laser module and same thing,nothing burning.  Have deleted lasergrbl and reloaded several times and dame thing, can not burn anything. I looked at the config files and I go not see lines 30,31,and 32.  Could this be the problem.  Im fairly new at his so take it slow and easy so I can understand. Help me pleae

StuartB4, 2019-11-21:
If you mean $30, $31 and $32 are not in the GRBL configuration list then you have an older version of GRBL,   may be version 0.8 or 0.9.  You will need to upgrade to version 1.1.
[image]

WHAAWHOO, 2019-11-21:
It does show the older version,  it is loading this everytime I download

closed 2020-01-27 by arkypita