arkypita/LaserGRBL #779: horizontal lines are not straight

GildDart, 2019-11-21:
Hello,
I have an Alfawise C30 engraver since a few days. I am using LaserGRBL that is realy great.
Every thing seems OK except that the horizontal lines are not straight. It looks like light waves or ripples.
It looks like a software issue (configuration ?) Do you have any idea ?
Thanks in advance for your help.
Gildart

arkypita, 2019-11-21:
Take a look here: https://github.com/arkypita/LaserGRBL/issues/30

arkypita, 2019-11-21:
And here

[image]

closed 2019-11-21 by arkypita

GildDart, 2019-11-21:
Thank you for your reply. I test many things regarding these topics and it's finaly a problem with the bottom wheel  of X axe...