arkypita/LaserGRBL #780: laser power suddenly dropped down

baselbaransi, 2019-11-21:
Hi All,
I was using The Laser and while the picture was carving and burning, suddenly the laser power went down and the picture didn't carve.

what would be the problem? i cant seem to find it.
[image]
 as you can see its on full power. i also tried setting it to S=1000 it also didn't help.

please help me.

StuartB4, 2019-11-22:
If the laser just stopped working it could be a bad connection somewhere, or the laser just died.
Although it says it's on full power on the screen, LaserGRBL wont know whether the laser is on or off.
It just sends the gcode to move the laser, the power settings for the laser are in the gcode.

baselbaransi, 2019-11-22:
The laser is on, but doesn't work on full power.
Ergo it doesn't burn the wood.
You mean the it could be a laser problem and i need to buy a new one?.
I only had it for a short time

arkypita, 2019-11-22:
Where did you buy your laser? What is the power of this laser? Can you post a link?
Maybe you can ask for refund or a new one.

Please take care that It is very common to find over-powered lasers sold on ebay/aliexpress.
I mean laser that mount 7W-8W diode, sold as 15W, and driven with more current of what the diode can handle continuously. The result is that these lasers seem very powerful, but they can't stay on for more than a few seconds without being damaged. The worst enemy of the laser is temperature. If they warm up above 50 ° their life drops drastically.

It is a bad practice, but unfortunately it is very common (the sellers on these platforms trying to do everything to attract buyers, selling products at a lower price than the others and cheating on the characteristics of the product). Only a few sellers are honest and report this information. Here an example:

[image]

This does not mean that it is not possible to use these lasers, but it is important to know that to prolong their life it is important to have some precautions:

Change the driver setting to give less current, **or**, work using half power (PWM 50%) 128 instead of 255 in S-Max, **or**, configure cooling cycle to pause job every few seconds, **and**, provide better heatsink and fan

closed 2020-01-27 by arkypita

LJg81, 2020-04-19:
@baselbaransi the very same just happened to me. About 20mins into a cut on full power. 7000mW laser purchased from Amazon..