arkypita/LaserGRBL #781: help

WHAAWHOO, 2019-11-22:
Getting frustrated, how do i update from version 0.9, everything I have read and tried does not work. Making my PTSD worse

StuartB4, 2019-11-22:
What board are you trying to upgrade?

WHAAWHOO, 2019-11-23:
[image: image.png]

This is the same board that I was using up till a week or so ago.  I was
working on a project and it just quit burning, I tried another laser module
and the same thing would not burn.  So I ordered this board an another
laser module, both x and y axis are working properly, the laser is shooting
a beam down just not burning.  I've looked at both computers that I use the
LaserGRBL on and now both of them say the version is 0.9.  So confused as
how it was working one minute and not the next,

StuartB4, 2019-11-23:
There's no image appearing on you post.

WHAAWHOO, 2019-11-24:
I can't seem to understand how to upgrade grbl to 1.1 it gets confusing and
I get frustrated easily but here the board

StuartB4, 2019-11-24:
Still no image. Are you clicking Attach files at the bottom of where you type in your replies.
If you are doing it from your phone it sometimes has problems uploading images here, I don't know why though.

StuartB4, 2019-11-24:
If you can't upload an image you can go on to google images and search for cnc controller board or laser engraver controller board and find an image the same as yours, then post a link to where you saw it.

WHAAWHOO, 2019-11-24:
https://www.amazon.com/gp/product/B07BMSH49Y/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

StuartB4, 2019-11-24:
Thats better.  This might help.
Goto here :http://wiki.sainsmart.com/index.php/101-60-280PRO#Firmware_upgrade
Look near the bottom for Firmware Upgrade and click the Firmware upgrade to 1.1 link, that will download the firmware. 

When it's downloaded extract the file in to a folder. It sholud look like this:

[image]

Then connect your machine and watch this video on youtube. It's only 2.5 minutes.
https://www.youtube.com/watch?v=2lNShEBYQ_0

WHAAWHOO, 2019-12-01:
thanks for the info, even though I havent got it to update, it stays on
uploading

closed 2020-12-16 by arkypita