arkypita/LaserGRBL #782: Help with my laser because i cant get it to work.

nichtAlex, 2019-11-22:
I bought a 6W laser from ebay. It has a +12V, GND, TTL and Temparatur cable. they ebay listing sometimes says its TTL controlled and sometimes its PWM. However its only compatible with a 3.3V signal. I have a logic level shifter which should fix this problem and convert the 5V PWM signal from the arduino to a 3.3V signal. however, my arduino doesnt seem to output anything on pin D11 whey i use either M3 or M4. Atleast my multimeter doesnt pick up anything. Also my logic level shifter is only marked really bad. instead of something like HV and LV connectors it only has VA and VB so i didnt figure out which one is which yet. If im not wrong i have to connect the laser +12V and GND to my power supply and the TTL/PWM pin to Arduino D11 but it doesnt seem to work.

StuartB4, 2019-11-22:
Try pin 12 for the pwm signal. It was switched in version 0.9 of grbl to pin 11, so you may have an 0.8 version which used pin 12. 
https://github.com/grbl/grbl/wiki/Connecting-Grbl

arkypita, 2019-11-23:
Instead, if you have v1.1, please note that if you have laser mode "on" ($32=1) it is normal that you dont have any output on PWM if you turn on the laser when the head is not moving.
Laser  is turned on only during linear movement.

Google "grbl laser mode" to read about it.

nichtAlex, 2019-11-23:
Oh. ok. Im using GRBL 1.1h. I didnt know that the pin only sends an output signal if the axis are moving. That most likely was the problem then. If I get all the remaining parts today I will mount the laser and then test it. I will reply to this thread again if it works / doesnt work. Thanks for your help.

nichtAlex, 2019-11-23:
Update: Laser now works. I added the Logic Level shifter and everything works. I had to connect the OE pin of the logic level shifter to one of the 3.3V out pins in order to get it running. no idea why tho. now it works. Im currently experimenting with the settings because cutting paper with 100% on a 6W laser is probably a bit overkill.

closed 2019-11-23 by nichtAlex

StuartB4, 2019-11-23:
Hi nichtAlex. 
Can you show me how you added the logic level shifter (pics if possible).
I've ordered a 7 watt laser from Aliexpress, probably the same as yours but different power.
It also uses 3.3 v for the PWM so I will need to add the shifter to mine. 
Someone sugessted I use a voltage divider and that should work but I would like to know how you did it because your's is working.
This is the one I ordered.
[image]
[image]

Thanks.
Stuart.

nichtAlex, 2019-11-23:
Mine looks almost identical.
This is how I connected the logic level shifter:
[image]
+5V & +3.3V are from the arduino
these are required to let the shifter know what the input voltage and the output voltage is.
GND is just connected to GND on the arduino
the 5V PWM is connected to Pin D11 on arduino (Laser PWM output)
the 3.3V PWM goes to the laser.
I had one problem which was that I had to hold a wire to the OE pin for the laser to work. I fixed that by simply sticking the other end of the wire through ond of the A pins.

I used B2 and A2 because they arent directly next to the input voltage pins so its a bit easier to solder wires there. Hope i could help you.

StuartB4, 2019-11-23:
That's brilliant, really helpful.
I'll definately try it when my laser arrives, could be another 20 to 25 days though from china to uk.
Thanks very much. :+1:

nichtAlex, 2019-11-23:
No problem.
If you're lucky it might arrive earlier. I bought mine on Novemebr 11th and it should have arrived between November 28th and December 10th. I got it on November 22nd.
I bought it on ebay but it said its from Shen Zhen, China. 11 days from China to Germany is really fast. Sometimes stuff takes a few months and sometimes only a few days...

StuartB4, 2019-11-24:
Yeah I know what ya mean. In the UK I may even have to pay customs import tax as well. 
I could end up paying more in tax than the laser cost.
It's really annoying.

StuartB4, 2019-11-24:
Hi  nichtAlex .
Just been looking at logic level shifters.
It seems the OE pin is an output enable pin. It needs to be held high to enable the output.
Which is why you had to link it to another pin.

Been looking at this one, it has a 10k (103) resistor linking the OE pin back to the board to hold it high so it works straight out of the box.

[image]