arkypita/LaserGRBL #783: Beta test for custom header/footer

arkypita, 2019-11-23:
Hi all!
I need beta testing this version: https://github.com/arkypita/LaserGRBL/releases/tag/v3.0.19
Anyone is welcome to download and use and give feedback about:

- customize header/footer/cycle codes
- realtime commands sent from custom button and header/footer/cycles

If possible also test the behaviour of macros inside header/footer.

Available macros:

macro | value
-- | --
left | Left margin of loaded image (X)
right | Right margin of loaded image (X)
top | Top margin of loaded image (Y)
bottom | Bottom margin of loaded image (Y)
width | Width of loaded image
height | Height of loaded image
jogstep | Actual value of jog step cursor
jogspeed | Actual value of jog speed cursor
WCO.X | Actual value of WCO X (Work Coordinate Offset) use .X .Y .Z
MPos.X | Actual Machine position. Support .X .Y .Z
WPos.X | Actual Work position. Support .X .Y .Z

Available realtime commands: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#grbl-v11-realtime-commands

distebia, 2019-11-23:
potrei inserire nel codice personalizzato il comando pause a ogni linea come discusso su https://github.com/arkypita/LaserGRBL/issues/698?

arkypita, 2019-11-23:
> potrei inserire nel codice personalizzato il comando pause a ogni linea come discusso su #698?

no, this feature is not intended to do that

B1Gt0ne, 2019-12-02:
How do I enter a code?
I want to add M452 P3 R255 F200 S1 as a header

arkypita, 2019-12-02:
Download [latest pre-release](https://github.com/arkypita/LaserGRBL/releases/download/v3.0.24/install.exe) 

open "grbl-settings" and edit this page

[image]

B1Gt0ne, 2019-12-02:
I have tried this but it makes no difference

arkypita, 2019-12-02:
Difference is not in exported gcode.
Exported GCode does not include custom header/footer

B1Gt0ne, 2019-12-02:
OK I see. Is this a possibility for a future release?

arkypita, 2019-12-03:
> OK I see. Is this a possibility for a future release?

I don't think that is correct to export header/footer into saved gcode.

Custom header/footer, and code between passes, is sent by LaserGRBL to the engrave at the act of start job (header), between each passes (passes), and after the last pass (footer).

If you put this code into export, when you reload the exported file in LaserGRBL you can't distinguish them from the rest of GCode, so they will be repeated twice (and worse, between each pass). 

Furthermore, the changes you make to the header / footer configuration would no longer apply to the files you exported (because they now have that information inside them).

For all these reasons I am convinced that it is not right to export these lines to the saved file.

closed 2020-01-27 by arkypita