arkypita/LaserGRBL #787: Visualisation Error

GrahamJB, 2019-11-23:
Just FYI.
Following GCode producxes a visualisation error, the second set of lines are above the first. Although the visualisation is incorrect while processing the file the blue+ follows the correct path.

$32=1
(TracingTest Gcode Generator-V2)
(Focus Test - Draw 11 lines with a Z axis step)
(Power for Lines 250, Feed 300)
(Power for reference marks 350, Feed 300)
(Line Height 10, Spacing 2, Z Step 1 mm)
G94
G17
G21
G0 G90 X0 Y0
G91
M3
F300
S350
(Draw Reference Marks)
G1 Y1
G0 X2 Y-1
G1 Y1
G0 X2 Y-1
G1 Y1
G0 X2 Y-1
G1 Y1
G0 X2 Y-1
G1 Y1
G0 X2 Y-2
G1 Y2
G0 X2 Y-1
G1 Y1
G0 X2 Y-1
G1 Y1
G0 X2 Y-1
G1 Y1
G0 X2 Y-1
G1 Y1
G0 X2 Y-1
G1 Y1
G0 G90 X0 Z0 Y2
G91
G0 Z-5
S250
G1 Y10
G0 X2 Y-10 Z1
G1 Y10
G0 X2 Y-10 Z1
G1 Y10
G0 X2 Y-10 Z1
G1 Y10
G0 X2 Y-10 Z1
G1 Y10
G0 X2 Y-10 Z1
G1 Y10
G0 X2 Y-10 Z1
G1 Y10
G0 X2 Y-10 Z1
G1 Y10
G0 X2 Y-10 Z1
G1 Y10
G0 X2 Y-10 Z1
G1 Y10
G0 X2 Y-10 Z1
G1 Y10
G0 G90 X0 Y0 Z0
M2
(END OF PROGRAM)

[image]

arkypita, 2019-11-23:
Damn! I have write LaserGRBL thinking that only one G code or M code is valid on a single line.
The issue is on line G0 G90 X0 Z0 Y2, where LaserGRBL parse the G0 and skip G90.

GrahamJB, 2019-11-24:
A small problem,

 

Personally I hate it when I do things like that so I sympathise.

 

Just remember that if design and programming were easy everyone would want to do it!

 

Graham Bland