arkypita/LaserGRBL #788: richiesta aiuto

Vespista, 2019-11-24:
Buongiorno, sono Giorgio, 
posso chiedere come fare la procedura e settaggio di laserGRBL completi e corretti per passaggio da laserato a penna per disegno su uscita servo.
Grazie

arkypita, 2019-11-27:
Ciao. LaserGRBL non è fatto per gestire una penna.
Se il tuo servo però si attiva con M3 e disattiva con M5 forse potresti farcela.

Ti basta togliere "Support PWM" nella pagina di configurazione di LaserGRBL e lui non creerà i comandi S che regolano la potenza, ma solo M3/M5.

Hi. LaserGRBL is not made to handle a pen.
If your servant is activated with M3 and deactivated with M5, maybe you could do it. You just need to remove "Support PWM" on the LaserGRBL configuration page and he will not create the S commands that regulate the power, but only M3 / M5.

Vespista, 2019-12-02:
I tried, but the servant does not return to zero, I have to try with a selenoide on off, I will continue to look in your posts very interesting and informative.
thank you so much for your help.
ho provato, però il servo non ritorna a zero, devo provare con un selenoide on off, continuerò a cercare nei Vostri post molto interessanti e istruttivi.
grazie mille per il tuo aiuto.

StuartB4, 2019-12-02:
This may interest you.
https://github.com/gnea/grbl/pull/245

Vespista, 2019-12-03:
buonasera,
si interessante avevo iniziato a tradurre, continuerò con più pazienza e cercando di capire.
GRAZIE.

closed 2020-01-27 by arkypita