arkypita/LaserGRBL #789: Compatability with LaserAxe

Dragonreborn88, 2019-11-24:
I have a Laseraxe DIY system and been having issues with their software (suddenly maxing out my CPU and freezing up the system) so I was looking for another potential software to use. Is there a driver or anything needed to connect/configure? I was able to connect the laser and jog it with the toggles but when I attempt to configure there is no information listed

StuartB4, 2019-11-24:
If you mean configure LaserGRBL, what happens when you click on the Grbl Configuration option in the Grbl menu top left? 
Normally you would configure the grbl settings from there, and also change any settings in the settings menu there.  If the configuration menu settings are empty it's because you have laseraxe firmware and not GRBL.
You change other settings after you have loaded your image in to it.

StuartB4, 2019-11-24:
Also you have to be connected to see config settings for grbl, if not connected it reads a cached version and which is not editable.

brakthehun, 2019-11-28:
What do you need to configure??

closed 2020-01-27 by arkypita