arkypita/LaserGRBL #794: Part of the picture disappeared when burned out

RomulRS, 2019-11-27:
Hello friends. Help understand what the problem is. Missing part of the picture in the middle.
[image]
[image]
[image]

arkypita, 2019-11-27:
This is really, really incredible. I have no idea how it could have happened.
I assume it is a hardware problem and only you can understand and tell us what it was. You should follow the process visually and figure out what jams when it passes through that area.

brakthehun, 2019-11-28:
Can you upload the cut file??

RomulRS, 2019-11-28:
> Can you upload the cut file??
Upload to? A laser engraver or a forum here?

> This is really, really incredible. I have no idea how it could have happened.
> I assume it is a hardware problem and only you can understand and tell us what it was. You should follow the process visually and figure out what jams when it passes through that area.
I observed that the laser in that place, which disappeared on one axis correctly, but on the other stood in place. It was as if voltage did not go to one motor for a while, or there was a program failure along the coordinates of one axis.
I noticed that the laser moved along one line for a long time back and forth. As if the voltage did not reach one engine for a while, or the program crashed along the coordinates of one axis.

arkypita, 2020-01-27:
Did you figured out what was the problem?  I am courious

closed 2020-12-16 by arkypita