arkypita/LaserGRBL #795: Communicating with ESP32 running GRBL

brakthehun, 2019-11-30:
I've had several issues communicating with an ESP32 running GRBL using LaserGRBL.

It seems -
When bluetooth is active - LaserGRBL seems to take a "BT echo" as a response during Gcode streaming.
LaserGRBL Randomly disconnects while idle or streaming. UGS and Bcnc do not disconnect.
I no longer see a verbose mode.

arkypita, 2019-11-30:
There is a command line that activate a logging of any tx/rx and communication event.

Get some log and we can try to understand

https://github.com/arkypita/LaserGRBL/search?q=is%3Aissue+comlog&type=Issues

In this

brakthehun, 2019-12-01:
[sessionlog.txt](https://github.com/arkypita/LaserGRBL/files/3907806/sessionlog.txt)
[image]
[image]
[image]
[image]

arkypita, 2019-12-01:
comlog.txt is more interesting that sessionlog.txt
try get it using indication you can find in threads i link you: https://github.com/arkypita/LaserGRBL/search?q=is%3Aissue+comlog&type=Issues

UnexpectedDisconnect is generated when LaserGRBL is reading from COM port and the read fails, not because there is no data but because the COM port is not in a readable state (usually because it was closed by the OS).

I have no specific log on it, but if needed I can add more logging in the two points where this error is throw, and give you a test version.

What I can suggest is to try to set "Syncronous" in streaming mode, and "Quiet" in threading mode.

arkypita, 2019-12-02:
latest pre-release allow to enable com logging via "?" menu option
https://github.com/arkypita/LaserGRBL/releases/tag/v3.0.24

brakthehun, 2019-12-04:
[comlog.txt](https://github.com/arkypita/LaserGRBL/files/3920673/comlog.txt)
[ExtendedLog.txt](https://github.com/arkypita/LaserGRBL/files/3920674/ExtendedLog.txt)

arkypita, 2019-12-05:
Unfortunately I can't detect useful information from the log to understand the problem you're highlighting.

> What I can suggest is to try to set "Syncronous" in streaming mode, and "Quiet" in threading mode.

No changes with this settings?
Have you ever had the chance to try a different PC?

arkypita, 2019-12-09:
If you want you can follow this instruction to record a log using a third-part com logger:
https://github.com/arkypita/LaserGRBL/issues/715#issuecomment-563354745

I can try to find the difference that make UGS not sensible to this issue.

arkypita, 2020-01-27:
Hi @brakthehun 
Did you fix this issue by yourself? Let me know

brakthehun, 2020-01-28:
The ESP32 UNO package doesn't crash with your program,, but the standard ESP32 package does. You can close this down. There is still more to figure out.

closed 2020-01-28 by arkypita