arkypita/LaserGRBL #797: M3 and M5 commands do not work

Polhmann, 2019-12-05:
Hi

A few days ago I bought the following CNC model:
https://es.aliexpress.com/item/32925976771.html?spm=a2g0s.9042311.0.0.4a5163c0YH4gVC

I have followed all the steps and the problem I have is that in the LaserGRBL program when I put "M3 S30" it does not turn on the laser.
The laser only works continuously if I give the power button, but its intensity is not regulated

StuartB4, 2019-12-05:
What version of grbl do you have, it should tell you here when you start LaserGRBL and click connect.
[image]

StuartB4, 2019-12-05:
If you have version 1.1 follow this:

[image]

arkypita, 2019-12-05:
If you have $32 laser mode enabled (=1) the laser does not turn on when not moving.

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

[image]

Polhmann, 2019-12-05:
Perfect! Problem solved! Many thanks!

Now my second problem is that in the images I upload, they appear as the other way around, where the line of the drawing is left unrecorded and the empty is recorded.

I'm so sorry but I'm a newbie in this

asinthe1, 2019-12-06:
Hi previously i use 2.5W laser which contain 3 pins (12v+,pwm,gnd) its works fine newly bought 15W laser which contains 4 pins (12v+,gnd,pwm+,pwm-) then i connect pwm- and gnd wire together and pwm+ to pwm wire in the board now its always on  anyone can help me to solve this issue

Smart-Stadium, 2020-01-14:
Hello, I have the same issue. My laser is continuously on. M3, M4 or M5 don't work. I followed all instructions but without result. I have GRBL 1.1f on a card: woodpecker cnc camtool v3.4.

When I tried an engraving. The move is synchronized between my CNC and LaserGRBL and I can see a variation of the light but nothing. Do you an idea to help me ?

closed 2020-01-27 by arkypita