arkypita/LaserGRBL #799: 8mm offset on x axis after homing problem

armature81, 2019-12-07:
Hi
I have problem on machine homing 
I download latest grbl v1.1h.20190825
When i make homing for the machine and start my work From zero i found there is x axis offset about 8mm
Look at the picture please for machine Coordinate after homing you can see this -8 on X axis 
So please any idea how to remove this offset?
[image]
[image]

StuartB4, 2019-12-07:
Have you checked $27 in GRBL config settings on the Grbl menu.

armature81, 2019-12-07:
But $27 will be applied on all axis not only x
Also pull off working good 
But this 8mm shift once start engraving

StuartB4, 2019-12-08:
Have you tried it with Hard Limits off. $21=0

closed 2020-01-27 by arkypita