arkypita/LaserGRBL #803: svg import positions misaligned

Skyfulrucm, 2019-12-12:
Hi, i am facing a weird issue.
Sometimes when i fill with eggbot or creating a svg in Inkscape, the Lasergrbl Software is not importing right.

My example of today is attached. It misalign some items.

Do you have an idea what is wrong?
[image]

Many thanks

Ludwig

rsalvarez1, 2019-12-19:
I have the same problem, in my case a 4 ft x 8 ft area  is converted in a 106758 x - 208921 , if i create in the 8 x 11 inch  space, i have to move the step/mm to cut and adjust size by test

arkypita, 2020-04-08:
Sorry for late reply. What version of Inkscape do you have?

arkypita, 2020-04-08:
It was reported that there are issue with SVG format generated by inkscape 0.92 when SVG generated by inkscape 0.91 is always good

https://github.com/arkypita/LaserGRBL/issues/872
https://github.com/arkypita/LaserGRBL/issues/907

Skyfulrucm, 2020-04-08:
Yes indeed, i am using version 0.92
Thanks for your reply, i will test it with version 0.91 - Hopefully i do not miss a feature :)

cheers Ludwig

closed 2020-04-08 by Skyfulrucm

arkypita, 2020-04-08:
Unfortunately SVG support is still under development in LaserGRBL, I'm looking at if I can correct those small problems that have been reported to me. Thread about SVG continues here: https://github.com/arkypita/LaserGRBL/issues/451

arkypita, 2020-04-08:
Do you still have the file that gave you problems? If I can zip it and pass it, I can test it.

Skyfulrucm, 2020-04-08:
Hi, no acutal not.
I'd moved every single item a litte from its origin position and kill all groupings - Then it was working.
May this help you in finding out what happened.

cheers

Ludwig