arkypita/LaserGRBL #804: laserGrbl con laser co2

distebia, 2019-12-14:
Salve, attualmente utilizzo Lasergrbl, arduino uno e un laser a diodo 450nm 10W (modulato in PWM) e riesco a fare foto su legno sia usando la conversione Linea per linea sia il Dithering 1bit BN. Se anzichè usare un laser a diodo vado a usare un laser co2 posso aspettarmi la stessa cosa? In sostanza l'alimentatore del laser e il tubo stesso mi permettono di lavorare in PWM con scala potenza da 1 a 255? Oppure sarà soltanto on/off? nelle specifiche dell'alimentatore leggo: 
La potenza è regolabile dal controllo TTL o da una variabile 5K
resistenza, design compatto con eccellente circuito di controllo PWM. esso
supporta tutti i laser CO2 fino a 40W.

arkypita, 2019-12-16:
Io non ne ho idea, ho solo laser a diodo. Prova a scrivere in inglese che magari qualcun'altro può risponderti. Prova anche sulla pagina facebook www.facebook.com/pg/LaserGRBL/community/

distebia, 2019-12-17:
Hi, I currently use Lasergrbl, an arduino one and a 450nm 10W diode laser (PWM modulated) and I can take pictures on wood either using Line by Line conversion or 1bit BN Dithering. If instead of using a diode laser I'm going to use a co2 laser, can I expect the same thing? Essentially the laser power supply and the tube itself allow me to work in PWM with power scale from 1 to 255? Or will it only be on / off? in the specifications of the power supply I read:
The power is adjustable by TTL control or by a 5K variable
resistance, compact design with excellent PWM control circuit. it
supports all CO2 lasers up to 40W.

e6910, 2021-03-14:
Sorry for the extremely late reply. It "should" work. I'm currently gathering parts to turn my engraver into a CO2 laser cutter. I'm planning on using the frame/rails and steppers, and an Arduino to control it all. You should connect the TTL input to the PWM pint on the Arduino.

HOWEVER, you should investigate... a LOT!!! This is not a small task to do. It is orders of magnitude more dangerous than using your current diode. Severe skin burns can (and will) happen if you are not careful. Or worse. These things WILL BLIND YOU. The whole thing should be covered before you even think of test firing it.

Another factor to consider, these tubes are water cooled. Also, they do operate at 20kV. water and high voltages don't like each other very much. A good grounding system is imperative.

Also, you should think about smoke and fumes in general. 

Do at your own risk. You should have a good knowledge in electronics, electric mains, GRBL and GCODE generation. 

If you don't know about that, I encourage you to consider buying a assembled, commercial unit.

Marino

closed 2023-09-09 by distebia