arkypita/LaserGRBL #805: problem with Y axis movement

tpres500, 2019-12-14:
Y axis will move same direction no matter which arrow you click in manual movement. Will go the same direction or wrong direction when G code is typed directly in. No on X axis, only Y.

StuartB4, 2019-12-15:
You must have something wrong with your machine. It works fine when pressing the arrow keys or typing in commands. Y up arrow moves Y axis away from home and Y down arrow moves it towards home, same when typing G0 Y10 or G0 Y-10.  May be you need to invert the Y axis in GRBL configuration settings.
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

tpres500, 2019-12-15:
The jog works fine unless you have it in continuous mode. When you switch to continuous, the Y axis moves the same direction no matter which arrow you click. It always moves towards 0. And, it will not home the Y axis when in continuous mode. I have tried with 2 different computers and does the same thing with both.

StuartB4, 2019-12-15:
It only works with grbl 1.1. and it needs to have $130, 131 and 132 in grbl configuration settings set to the actual size of the engraving area. Which version do you have ?

tpres500, 2019-12-16:
I downloaded the latest stable version. It said "Include all changes of the last pre-release (3.0.19 to 3.0.23) plus some minor changes." I have the max area set in the config.

StuartB4, 2019-12-16:
So you have version 1.1 of GRBL already.

tpres500, 2019-12-16:
I don't know. I just downloaded what is at https://github.com/arkypita/LaserGRBL/releases/tag/v3.0.24

StuartB4, 2019-12-16:
Start LaserGRBL and click Connect and it will tell you here:
[image]

arkypita, 2019-12-16:
Are you sure that it's not an electrical problem?

Y direction is outputed by arduino pin 6

[image]

This pin goes to motor driver pin "DIR" throug the pcb that connect arduino to the motor driver. The motor driver generate pulses to control the motor, in accord to direction/step

[image]

Maybe you can have:
- a death arduino pin
- a shortcut on the pcb path
- an interrupted pcb path
- a death motor driver

Did you have a multimeter? Have you already tested this part?

arkypita, 2019-12-16:
If your board use arduino nano those are the connections:

[image]

tpres500, 2019-12-16:
I do have 1.1. I actually cut parts with the software last night so everything worked fine doing that. I did more experimenting today and the Y axis did not act up. I switched it back and forth from incremental to continuous and it worked as it should. However, I swear that it did not in previous testing. As I stated, it would go the same direction no matter which arrow you clicked. The position display showed it going the same direction as well. It wasn't like the software was telling it do one thing and the machine was doing something else. The machine was doing what the software was telling it to. Sorry for the bringing this up in the beginning but it did do as I said and for whatever reason it doesn't do it now. Thanks for the replies.

StuartB4, 2019-12-16:
Ok, that's good. No problem. :+1:

arkypita, 2019-12-16:
Ok, so it looks like it solved by itself? If it happens again you may make a short video that takes up the whole LaserGRBL screen while the system behaves this way.

I'm curious to see what commands generate LaserGRBL for jogging, and how grbl responds with positions and movements.

I want to understand if it is LaserGRBL that generates the wrong commands, or if it is grbl that executes them wrong.

arkypita, 2019-12-16:
Please note that continuous jog need to know the exact size of table to produce the correct jog commands.

LaserGRBL can read this info only if you edit/show them from "grbl - > grbl configuration" menu. Maybe you have simply opened the window and LaserGRBL read that values from the machine for the first time? Engraver need to be in "idle" state for LaserGRBL to be able to read that values.

tpres500, 2019-12-16:
I put in 4000 mm for each axis for max travel for soft limits even though the machine has a larger area than that. In testing I was making smaller items so I didn't care if it knew the full extents of the machine nor could I see a reason that it needed to know a long as I wasn't exceeding them during program execution. I will put in the actual max readings when I get a chance to jog it to those limits. Since I have just gotten the steps/mm adjusted so the parts are the proper size, I can now jog the machine from 0,0 to the max and know what the limits actually are.

closed 2020-01-27 by arkypita