arkypita/LaserGRBL #806: abort still leaves last move to be completed

tpres500, 2019-12-14:
even in syncho mode, it's impossible to stop the machine and have the last commanded move completely cancel. It stills holds the last command in the buffer and there's no way to purge it. The only workaround is to shut off the controller and then let grbl run again until the last commanded move is finished. That loses position. The software needs to purge whatever move was last commanded and allow the machine to stop in place at the last position.

brakthehun, 2019-12-20:
What version GRBL Firmware is your controller running??

tpres500, 2019-12-20:
It's the firmware that came with benbox. I tried to flash the lasergrbl firmware but it said it couldn't do it.

brakthehun, 2019-12-20:
https://www.youtube.com/watch?v=9AecpJBzJDM

I found this. I'll look up what the actual controller is later when I get to my office.

tpres500, 2019-12-20:
I downloaded x-loader and will try to flash the controller with it. It would not flash from within lasergrbl.

tpres500, 2019-12-20:
I was able to flash the controller with the lasergrbl firmware. I will do some testing to see if it changed anything. Thanks,

closed 2020-12-16 by arkypita