arkypita/LaserGRBL #811: Lasergrbl

laserLee, 2019-12-21:
Hi I was use lasergrbl it was working fine then when I put the laser on of mode on now it seems to be stuck in some sort of mode instead of scanning can anyone help me please

StuartB4, 2019-12-23:
Can you give us a bit mor information please, and any pictures of what is happening.

laserLee, 2019-12-23:
I fixed it now thanks very much I had to change $32=1 and that fix it

Is there a way to of get a center start point and a way to see what the
laser will do like on t2 laser

StuartB4, 2019-12-23:
Not sure about the center start point but you can add a custom button that draws a box with the laser on low power around the image so you know where it will start burning on the workpiece.

Read this:
Function: Trace a box around the image you have prepared so you
know exactly where on the workpiece the laser will go when burning.
Type: Button. - Single press.
Make sure you are CONNECTED and your image is prepared for burning, then use the button.

M4 is Laser On.
S2 sets the laser on low power. This can be any number between 1 and 255.
1 = lowest power, 255 = highest power.

F500 is the speed the laser will move around the box, you can change this
to a higher or lower number to increase or decrease the speed.
M5 is Laser off when completed.

Copy  the code below starting with M4 down to M5 and paste it in to GCode box.

M4 
S2
F500
G0 X[left] Y[bottom] 
G1 X[right]
G1 Y[top]
G1 X[left] 
G1 Y[bottom] 
G0 X0 Y0 F500
M5

Tool Tip: Framing - Laser On Low Power. S2. Speed 500

Enabled: Idle Program.

RIGHT CLICK on an empty part of the bar and click Add Custom Button.

[image]

laserLee, 2019-12-23:
Thanks very much thats great can I set 2 custom button how do I set laser
to come on and of is it m4 $2 me $0

StuartB4, 2019-12-23:
Custom button for laser on, press again for laser off.

Function: Switch Laser on low power for focussing.
Type: TwoStateButton.
Click On - Click Off.
Make sure you are connected then use the button.
This button is only valid if you have GRBL 1.1.

$32=0 switches off laser mode. $32=1 switches laser mode back on.
M4 = Laser on. M5 = Laser off. S = Laser power setting.

GCode box 1:

$32=0
M4
S3

GCode box 2:

$32=1
M5

Tool Tip: Laser ON (Laser Mode=OFF) / Laser OFF (Laser Mode=ON)

Enabled: Idle.

laserLee, 2019-12-23:
Thanks very much mate that's amazing thanks for all your help will let you
no how I get on

StuartB4, 2019-12-23:
No problem. There is also one for finding center of the prepared image, but when you switch it off it will return to the home position and start burning at the X0,Y0 position, not from the center.

Function: Find center of prepared image.
Click On - Click Off.
Type: TwoStateButton.
Make sure you are connected and your image is prepared for burning, then use the button.
This button is only valid if you have GRBL 1.1.

GCode box 1:

$32=0
G0 X[left+width/2]
Y[bottom+height/2]
G1 M4 S5 F900

GCode box 2:

$32=1
G0 X0 Y0
M5

Tool Tip: Find Center Of Image. Click On - Click Off.

Enabled: Idle Program.

laserLee, 2019-12-23:
Thanks very much mate that's great

StuartB4, 2019-12-23:
You may never to, but if you do for some reason need to just flash the laser on and off you can use this one.

Function: Flash Laser.
Type: PushButton. 
Press and release to flash laser on and off. 
Make sure you are connected and then use the button.
This button is only valid if you have GRBL 1.1.

GCode box 1:

$32=0
M4 S3

GCode box 2:

$32=1
M5

Tool Tip: Press and Release To Flash Laser.

Enabled: Idle.

laserLee, 2019-12-23:
Thanks mate that might come in handy I do need to do that from time to time

StuartB4, 2019-12-23:
No problem. :+1:

laserLee, 2019-12-23:
I am uploading an image from gimp and the size is coming out all wrong if I
load an image 100mm by 100mm then lasergrbl recognise the size but it only
comes out half that 50mm by 50mm what would cause that any idea mate

StuartB4, 2019-12-23:
Can you upload a picture of your GRBL configuration settings. 
You can do it by starting LaserGRBL and click Connect then goto Grbl menu in and click Grbl Configuration.
When the list of settings pops up stretch the window to include all the $ settings.
Get a screen shot of it or a photo from your phone as clear as possible and post it here.
Thanks.

appieollo, 2019-12-24:
I had the same problem with the size. I changed the steps from 40 to 80 and now have the right size…. just as simple as that for me ;-)

closed 2020-01-27 by arkypita