arkypita/LaserGRBL #812: Gcode issue

B1Gt0ne, 2019-12-21:
In previous versions of LaserGRBL the gcode outputted as follows:
M4
G1 X33.362 Y6.191
G1 X33.416 Y6.45
G1 X33.487 Y6.832
G1 X33.567 Y7.3
In the latest version the output is as follows:
M4
X33.362 Y6.191
X33.416 Y6.45
X33.487 Y6.832
X33.567 Y7.3
My machine requires the G1 on each line to work.

Any ideas please?

guillaume-rico, 2020-01-17:
There is a change in the code (commit made 2018/10/17) : GCode is now compressed by avoiding sending again same G-Nr and unchanged coordinates .
You can change this settings but you need to compil LaserGRBL. Download code in github, in the file gcodeRelated.cs (folder SvgConverter), change the line : 
`private static bool gcodeCompress = true;`
by
`private static bool gcodeCompress = false;`
Download visual studio (with .net module) and compil

DaGweeds, 2021-10-15:
@guillaume-rico 
I changed the line as you stated and recompiled it, but still I still get compressed code:

G1 X99 Y54.6 S0
G1 X99.2 Y54.4 S124
X99.4 Y54.2 S274
X99.6 Y54 S258
S0
G1 X99.8 Y54 S0
G1 X99.6 Y54.2 S240
X99.4 Y54.4 S285
X99.2 Y54.6 S226
X99 Y54.8 S233
S0

My Marlin based engraver does not support this and requires:

G1 X99 Y54.6 S0
G1 X99.2 Y54.4 S124
G1X99.4 Y54.2 S274
G1X99.6 Y54 S258
S0
G1 X99.8 Y54 S0
G1 X99.6 Y54.2 S240
G1 X99.4 Y54.4 S285
G1 X99.2 Y54.6 S226
G1 X99 Y54.8 S233
S0