arkypita/LaserGRBL #814: Laser power suddenly dropped off

baselbaransi, 2019-12-24:
Hello all,
While i was using the laser to engrave a picture, the laser power suddenly dropped off and it doesn't burn the wood any more.
What is wired that in the configuration the value of the laser power didn't change.
And there is still a laser light but so weak it doesn't burn the wood.
What could possibly be the problem and how to fix it.

Please help.
Thanks in advance

cowich, 2019-12-30:
Hallo baselbaransi,

i have the same problem, is there a solution?

StuartB4, 2019-12-30:
When a laser diode burns out or stops lasing it usually ends up with a weak light, like an led light. 
Still showing a light but not burning.
Hard to say without knowing anything about your machine, laser, settings etc.

baselbaransi, 2019-12-30:
Is there a way to fix it?

This is the laser

[image]

cowich, 2019-12-30:
Thanks for the answer StuartB4 - I have the  Ortur Laser Master 15 W Desktop Laser and have only been using it for 3 days - have lasered a lot and since today it has not burned anymore. the laser cannot have had its day after 3 days????

StuartB4, 2019-12-30:
cowich, if you've only had it a few day you should contact the seller. Not much else you can do really If you've checked all the connections and everything. If it just stopped burning part way through a job then it probably is a broken laser, or a power supply fault.

cowich, 2019-12-30:
I think so, thanks StuartB4

baselbaransi, 2020-01-01:
Do you think that this problem is fixable at a technician?

closed 2020-01-27 by arkypita

jnweiger, 2021-08-01:
I found this issue much later, while debugging exactly the same problem, LED presumably dead. Surprise, a replacement also did not work.

So just for the records: In our case, it was a quite profane connector issue. There is a small PCB at the left of the x-axis gantry, where one of the 4 wires goes to the end stop, and the other three continue to the LED head:

[image]