arkypita/LaserGRBL #816: LaserGRBL does not connect, putty or candle does

DetAtHome, 2019-12-25:
Hi there,
I just enhanced my chinese cnc mill with a laser module, and am stuck (no laser output)
So I wanted to give laser grbl a chance, but it does not connect. Putty or Candle do though.
I am on a woodpecker board running grbl0.9j
When I try to connect with lasergrbl the spindle immediatly starts spinning, but after a while I am disconnected again.
Do I need to update to a newer grbl version on the board (docs say lasergrbl is still compatible with grbl0.9j?
Clueless
Det

DetAtHome, 2019-12-25:
Ok, first I found a bug in my wiring - though the behavior is still strange. I was able to get the laser to fire up, but as I want some of the GRBL1.1 features (Laser off on G0 commands) I changed to grbl1.1f now. The board that I am using now which has a 3pin laseroutput (Woodpecker 3.2) does show the same behaviour: LaserGRBL does not connect, putty, candle, laserweb all do. But it seems I still have a wiring issue: Laser fan is always on, Spindle is always on, Laser is focused but not burning even through a sheet of paper (I guess the spindle drains all the watts...) 
As my laser module is a 4pin I guess I have done something hooked up wrong. 4th pin is temp I guess. I put that to gnd which might be wrong. 
If by any chance someone has an idea on what I did wrong any ideas are welcome, I guess the issue is rather in my wiring then with laserGRBL so feel free to close the issue right away

arkypita, 2020-01-27:
Did you try change soft-reset / hard-reset checkboxes in lasergrbl?

[image]

closed 2020-12-16 by arkypita