arkypita/LaserGRBL #817: Connect Arduino Laser Modules

fcgfilho, 2019-12-27:
How to connect these two laser modules with arduino uno through CNC Shield?

[image]
[image]

StuartB4, 2019-12-28:
Which CNC Shield?

fcgfilho, 2019-12-28:
CNC Shield V3.

[image]

StuartB4, 2019-12-29:
Sorry for the delay. I can only find a board with the  2 pin connectors on each end.

But it looks like you only need to use 2 of the pins on the 3 pin connector on your board.

If you look at this picture you can see that the power for the laser goes to 12+ and - on the bottom right socket and leave the other pin disconnected. Switch the switch over to TTL.

Then connect the + on the socket above it to the Z- or Z+ on the cnc shield,  and connect the negative- from that socket to one of the ground pins in the black row of pins on the shield, all the black ones are ground. 

It does'nt matter which Z pin you use they are both connected together on the circuit underneath the board.

[image]

fcgfilho, 2019-12-29:
Obrigado

closed 2020-01-27 by arkypita