arkypita/LaserGRBL #823: M4 command missing

OllieBFG, 2019-12-30:
I am running LaserGRBL v3.0.24 with a small table laser engraver. The engraving is burning the corners, as the software is using M3 as ON command. I read that I need to run the M4 command, but the software does not give me this option. How can I add the M4 command to the drop-down list?
[image]

StuartB4, 2019-12-30:
Start LaserGRBL and click Connect. Then click the Grbl menu and click Grbl Configuration.
Take a screenshot and post here.
Or just have a look at them and see if you have the 3 settings I've highlighted below.
If you do have them change the values to match the ones in the image and then click the Write button to store them. 
If you don't have them you will need to upgrade GRBL to a newer version.
[image]

OllieBFG, 2019-12-30:
Hi there,

 

Thank you very much for your fast response. 

 

I was able to change the settings according to your instructions and I now managed to get a clear engraving without burned edges. Great result.

 

Thanks again and best regards,

 

Oliver

 

P.S.: Happy New Year 😊 

 

 <https://user-images.githubusercontent.com/16980026/71599947-eb14c980-2b44-11ea-8124-acafdea47882.jpg>

StuartB4, 2019-12-30:
Thats good to know. Happy to help.
Happy New Year to you too. :+1:  :)

closed 2020-01-27 by arkypita