arkypita/LaserGRBL #824: New to this! driving motors?

ralphxyz, 2019-12-30:
When I first opened LaserGRBL it ran my motors.

Now I can not get it to drive the motors.

I loaded a .jpg file in LaserGRBL and it runs but my motors are not driven.

Thanks for the help.

ralphxyz, 2019-12-30:
Also I get nothing from the jog buttons!

StuartB4, 2019-12-30:
What machine do you have?

ralphxyz, 2019-12-31:
No "machine" just Arduino with GRBLcnc shield and a couple of stepper motors.

StuartB4, 2019-12-31:
Like this one? 
Is there any movement at all ?
Is the power supply ok?
There is a fuse on the cnc shield if it's one of these, circled in white, is it ok?
[image]

StuartB4, 2019-12-31:
Also, did you try resetting the Arduino via the reset switch on the shield?

ralphxyz, 2019-12-31:
Thanks for the replies. yes that is the cnc shield.

The fuse is good!.

Yes I did a reset from the shield

StuartB4, 2019-12-31:
Can you post a screen shot of your grbl settings.

[image]

[image]

ralphxyz, 2019-12-31:
Settings:
[image: LaserGRBL Settings 1.jpg]

ralphxyz, 2019-12-31:
Well that didn't work,let me try again. There isn't any image  options. How did you post the image?
I will try a link.

[Settings:]
(https://photos.app.goo.gl/i2UANPeeGf5bSp7cA)

ralphxyz, 2019-12-31:
Strange, I reloaded the GRBL sketch and now it works again!!

Don't know why but glad it works!

Now I'll be able to start asking questions about actually using LaserGRBL.

StuartB4, 2019-12-31:
Hi Ralph. Sorry for late reply, got a bit busy today. Glad you got it sorted.

One thing I would mention is to change $32 to 1 in the grbl configuration settings.
That enables Laser mode and gives you better results.

Also $130 and $131. change them to the actual engraving area/size of your machine.
For example if your machine can engrave an area of say 300mm in the X and 400mm in the Y direction, change the $130 to 300 and $131 to 400. It's just in case you want to use Continuous Jogging in the LaserGRBL's Settings menu. Which means it will jog as long as you keep the button pressed, but will stop automatically when you let go or when it reaches them numbers. So you can't over do it.

StuartB4, 2019-12-31:
To post an image just click on "Attach files by dragging & dropping, selecting or pasting them"  at the bottom of the comment box.

StuartB4, 2019-12-31:
If your motors/axis's move to slow change $110 and 111 to a higher number something like 10000 to 12000. 
If they take too long to get up to speed change $120 and 121, thats the (acceleration speed) to a higher number, not too high though or you will loose steps.

StuartB4, 2019-12-31:
To change any of the above settings just start LaserGRBL and click Connect.
Click the Grbl menu and select Grbl Configuration like you did to get the settings window.
Click in the Value box for the $ number you want to change and click again to enable edit.
Change the number to the one you want and press Enter afterwards, do this for all the settings you want to change and press Enter after each one.
When you've done them all click on the Write button at the bottom to store them in the Arduino. Then Close.

ralphxyz, 2019-12-31:
Thanks Stuart, next I need to build the machine I have a lot of things in
mind plus I have a K40 laser
with it's original controller I might substitute the GRBL CNC,

Got a lot to learn.

Thanks again,

Ralph

StuartB4, 2019-12-31:
I'm sure you'll enjoy it. Good luck with it, and there's always someone here to help out if needed.
Also on the LaserGRBL Facebook page here: https://www.facebook.com/groups/486886768471991

closed 2020-01-27 by arkypita