arkypita/LaserGRBL #825: Request: Preview bounds

ClutchplateDude, 2020-01-02:
It would be useful to be able to have the app make the laser travel around the bounds of the cut about to be executed to verify the positioning of the work piece. Set the laser to lowest power (S1?) and trace the square at full speed....

StuartB4, 2020-01-02:
Right click on an empty part of the bar where the buttons are and click on Add Custom Button.
[image]

The read this:

Function: Trace a box around the image you have prepared so you
know exactly where on the workpiece the laser will go when burning.
Type: Button. - Single press.

M4 is Laser On.
S2 sets the laser on low power. This can be any number between 1 and 255.
1 = lowest power, 255 = highest power.

F500 is the speed the laser will move around the box, you can change this
to a higher or lower number to increase or decrease the speed.

M5 is Laser off when completed.

Add this code to the GCode box and click Save.

M4 
S2
F500
G0 X[left] Y[bottom] 
G1 X[right]
G1 Y[top]
G1 X[left] 
G1 Y[bottom] 
G0 X0 Y0 F500
M5

Tool Tip: Framing - Laser On Low Power. S2. Speed 500

Make sure you are connected and your image is prepared for burning, then use the button.

Enabled: Idle Program.

ClutchplateDude, 2020-01-02:
Oh, wow, that was easy. Thanks so much for showing me how to do this.

StuartB4, 2020-01-02:
No problem, happy to help.

StuartB4, 2020-01-02:
I have 2 more that you may find useful.
This one.

-----------------------------------------------------------------------

Function: Switch Laser on low power for focussing.
Type: TwoStateButton.
Click On - Click Off.
Make sure you are connected then use the button.
This button is only valid if you have GRBL 1.1.

$32=0 switches off laser mode. $32=1 switches laser mode back on.
M4 = Laser on. M5 = Laser off. S = Laser power setting.

GCode box 1:

$32=0
M4
S2

GCode box 2:

$32=1
M5

Tool Tip: Laser ON (Laser Mode=OFF) / Laser OFF (Laser Mode=ON)

Enabled: Idle.

-----------------------------------------------------------------------

And this one:

-----------------------------------------------------------------------

Function: Find center of prepared image.
Click On - Click Off.
Type: TwoStateButton.
Make sure you are connected and your image is prepared for burning, then use the button.
This button is only valid if you have GRBL 1.1.

GCode box 1:

$32=0
G0 X[left+width/2]
Y[bottom+height/2]
G1 M4 S2 F900

GCode box 2:

$32=1
G0 X0 Y0
M5

Tool Tip: Find Center Of Image. Click On - Click Off.

Enabled: Idle Program.

-----------------------------------------------------------------------

closed 2020-01-27 by arkypita