arkypita/LaserGRBL #827: New Laser. no focus button

Mlengal, 2020-01-02:
Just installed LaserGRBL and notice there is no but to focus the laser on object.  I only have reset, homing, unlock

StuartB4, 2020-01-02:
Right click on an empty part of the bar at the bottom of the main window and click Add Custom Button.

[image]

Then read this:

-----------------------------------------------------------------------

Function: Switch Laser on low power for focussing.
Type: TwoStateButton.
Click On - Click Off.
Make sure you are connected then use the button.
This button is only valid if you have GRBL 1.1.

$32=0 switches off laser mode. $32=1 switches laser mode back on.
M4 = Laser on. M5 = Laser off. S = Laser power setting.

GCode box 1:

$32=0
M4
S3

GCode box 2:

$32=1
M5

Tool Tip: Laser ON (Laser Mode=OFF) / Laser OFF (Laser Mode=ON)

Enabled: Idle.

-----------------------------------------------------------------------

Mlengal, 2020-01-02:
Thanks,  When I do this I dont the same image as you have?

Mlengal, 2020-01-02:
I figured out the clip art  thanks

DonLeon13, 2020-01-10:
Hy i have also a problem with the Focus.
i have createt the Button as listet, but i dont see a laser dot to setup the Focus.
when i load a picture and start engraving the laser goes on an make the picture but when i try to start the laser with the Command it dont go on.
what can i do ?

StuartB4, 2020-01-10:
Try changing the M4 in GCode box 1 to M3 and click save. Then try it.

closed 2020-01-27 by arkypita

kipbits, 2020-05-11:
hey, same problem here, glad to finally get aiming/focus working. i noticed the fan doesn't turn on though... I'd like to manually burn 4 corners of my "print-bed", but I don't want to damage my laser. ...i guess I could use a low power mode + a marker although I'd like to get this as accurate as I can. is there a command to toggle on&off my fan too?

StuartB4, 2020-05-11:
Is your fan fitted to the laser or a seperate one.
If it's fitted it should come on automatically, usually when you switch the engraver on.

kipbits, 2020-05-11:
oh hey, it's mounted on the laser. I can confirm it spins while engraving, i'm hesitant to type in
S128 though and see if it explodes

kipbits, 2020-05-11:
i did try S30 and still no fan. don't really want to press the limit, although I am trying to burn the table...

StuartB4, 2020-05-11:
May be it's temperature controlled and only comes on when it gets too warm.

StuartB4, 2020-05-11:
What laser is it, did it come with the machine or have you bought it separately?

kipbits, 2020-05-11:
pretty sure it came with the kit(although it set on my shelf for a good year++ with all my other junk so who really knows). it's a 405nm 500mW and turns the fan on pretty quickly when I press play on a file. it's not turning on though now(neither M4 or M3)
[image]

kipbits, 2020-05-11:
...i noticed there is an F500 command whenever i start a file, although it doesn't seem to do anything now. ...maybe I will try S50...

kipbits, 2020-05-11:
i think you are right about the thermocontrols; just ran a test file on S30 ...No Fan.

StuartB4, 2020-05-11:
The F500 is the engraving speed usually.
Also, I would'nt worry too much about the fan. If the laser is only 500 mW that's only half a watt.
It will barely get warm in that big heatsink it's fitted to.
I run a 2 watt for 7 or 8 hours at a time without a fan.  Anything above that I would use a fan.

kipbits, 2020-05-11:
I finally just did it:
$32=0
M3
S63
...the fan turns on automatically! Thanks! ...and nothing blew up!

...the hole it burned is very tiny. i think i might need to mark it with a pen. ;)

StuartB4, 2020-05-11:
Thats good to know. :+1:  It's good when things don't blow up. :)

kipbits, 2020-05-11:
I suppose that doesn't mean someone else's wont. You asked the right questions. f.y.i. this is project converting from bangood/benbox.

StuartB4, 2020-05-11:
Good idea converting from Benbox. I've seen a lot of people complain about it,  but then tried to convert it and had more problems than when they started.
If it works for you that's brilliant.

kipbits, 2020-05-11:
yeah it was relatively cheap. and apparently supports pwm, so I thought it was pretty good deal(⌐■_■) ...though it did take a bit of fooling around; i'd like to post a tutorial some time, when life is not so hectic.