arkypita/LaserGRBL #828: trying to setup alfawise C10 3018 engraver

quotelawrence, 2020-01-03:
I am very curious if there would be any assistance at helping to instructing me on the setting 
I have been using the laser program which came with the unit DIY8000 but it didn't load properly on my laptop so a big portion of the control panel was unusable and could not be accessed so I downloaded your software it loaded excellently on both laptop and desktop though I am having trouble getting the machine to function with the program can you help

StuartB4, 2020-01-03:
What are the problems?

quotelawrence, 2020-01-03:
thank you and sorry for delay i uploaded the program and plugged the alfa in but it seems that the program doesn't recognize the alfa i have went into the alfa folder found the files there, and downloaded an image it wrote a program yet never activated the laser the program said it was rnning but the laser just sat there doing nothing

StuartB4, 2020-01-03:
Does it connect ok when you click Connect. 
Can you go in to the Grbl menu at the top and click Grbl Configuration.
Then get a screenshot of the settings and post it here.
To upload the screenshot here just click at the bottom of the comment window where it says Attach files and select your image.

quotelawrence, 2020-01-03:
Hi Stuart yes it said it was connected and I did import the grbl settings from the other system, I am currently at work so it will be later before I can take the screen shot and post thanks again

StuartB4, 2020-01-03:
Ok, no problem.

quotelawrence, 2020-01-04:
good evening got very busy but good news I switched a few setting and if began to operate the table travel has to be changed as the zero wants to run opposite the direction I once had it but maybe I will just start resetting the zero I need to do homework on creating buttons I wanted a few I saw in the demonstrations you indicate on the web cite but I wanted to possible have a low beam adjustment mode and a button for 40%, 50% 60%, just to see if that worked well or I need to learn where in the operating system the heat is designated thanks again I hope to be up and running very soon,

StuartB4, 2020-01-04:
Ok. 
Let me know how you get on.

quotelawrence, 2020-01-05:
good morning Stuart I have eliminated many issues and it is working well, I have added a focussing button, and have learned how to edit my programs to start having first travelled to the center of the target, then the program goes to the first command, these seem to be working well though I notice the software always travels from right to left having wasted travel time is there a setting where the machine actually starts but rather than travelling all the way back to the origin it merely drops to the next line and burns in the the opposite direction also to end this wasted back and forth travelling

quotelawrence, 2020-01-05:
Stuart I have also noticed my laser though I am told 255 is the max I have taken back to 15, 20, it still has power to burn does running these laser modules on less power lengthen their lives ?

StuartB4, 2020-01-05:
Hi. Glad to see you are getting things working.
Yes burning on lower power will prolong the lasers life. 
As for only burning in only on direction, have a look in settings for this:

[image]

quotelawrence, 2020-01-05:
Stuart thanks and yes I do have it checked I will uncheck and see if it speeds it up a bit I also added a cooling process of 30 seconds though I may increase thanks again I will let you know of the progress

StuartB4, 2020-01-05:
Ok thanks.

quotelawrence, 2020-01-05:
Hello again Stuart I was lasering another wooden engraving when I heard the machine take another time out cooling I looked in the room and noticed it was taking the time out cooling procedure with the laser still active burning sitting in one spot, is there a way of not repeating this issue

quotelawrence, 2020-01-05:
this is where is stopped to cool and continued to burn 

[image]
[image]
[image]
[image]

StuartB4, 2020-01-05:
The only thing I can think of is that you have GRBL 0.9 or earlier. There is a warning on the settings menu where the cooling settings are about it only working with GRBL 1.1.

Start LaserGRBL and click connect and see what it says here:
[image]

quotelawrence, 2020-01-07:
Hi Stuart you were correct I did in fact have the setting at the wrong level now I am having another issue my laptop and my desktop are not displaying the same GRBL setting so I took some screen shots to see if you may know what it is the is stopping the desktop from listing $32= spindle speeds or laser strengths 
[image]
[image]

StuartB4, 2020-01-07:
The only reason it would not show $30, $31 an $32 is if you were using GRBL 0.9 or earlier / older.
If you have GRBL 1.1 and you just unplug the machine from your laptop and plug it in to your desktop and start LaserGRBL and click connect they should both show the same. GRBL is on the machine controller board so it should'nt be different from one computer to another. Are you sure you have GRBL 1.1 ?

StuartB4, 2020-01-07:
[image]

Is this what you get when you click Connect?  Grbl 1.1  the h after the 1.1 does'nt matter it's just the latest version letter. It could be e, f,  g, or h.

quotelawrence, 2020-01-08:
Hi stuart sorry for so many issues though for some reason possibly there are different versions on laptop verses desktop 
[image]

quotelawrence, 2020-01-08:
yes stuart my desktop was Grbl .9

quotelawrence, 2020-01-08:
I have downloaded and tried repeatedly to get the 1.1h to load I am not sure of the settings the I had on my laptop but it took the 1.1, where my desktop seems to be stuck on .9, I may have to remove the software and start from scratch

StuartB4, 2020-01-08:
Thats strange. Because GRBL is on a chip on the engravers controller, when you unplug in and plug it in to any other computer it should show the same version 1.1  in LaserGRBL.

quotelawrence, 2020-01-08:
so Stuart maybe I did not mention I have 2 machines the first one I purchased and it was a counterfeit alfawise spinoff, so I complained they gave me a deal on a replacement and between the 2 I possibly have a problem with different boards chips grbl .9 verses grbl 1.1,
I cleared my system and reloaded the software and it loaded the same version .9 is there any changes I can make or is this determined on the hardware 
[image]

StuartB4, 2020-01-08:
Can you post a picture of the main control board on the one that has 0.9j on.

quotelawrence, 2020-01-08:
yes here it is i recently ordered another board though these orders from china take a month than i received something different 

[image]

StuartB4, 2020-01-08:
Thanks. 
You would need to change the firmware in the one that has 0.9j.
That is stored on the main chip on the control board. Or on the Blue Nano board on the controller on some boards. 
You can usually upgrade to 1.1h but it may not work properly after.

If the one with grbl 0.9j had the blue Nano board plugged in to the main board you could just get another Nano board and load GRBL 1.1h on to that, but keep the original in case it did'nt work.
Then just swap them over, they only plug in. 

You could do that by downloading the GRBL 1.1h hex file from here:  https://github.com/gnea/grbl/releases/download/v1.1h.20190825/grbl_v1.1h.20190825.hex

Then use a little program called XLoader to install the hex file from here:  http://www.hobbytronics.co.uk/download/XLoader.zip

Or you could download T2Laser Trial version from here: https://t2laser.wordpress.com/download
There is a built in option to upgrade grbl to 1.1.

I would'nt like to tell you which way to do it in case it causes more problems if it does'nt work properly.

quotelawrence, 2020-01-08:
Stuart you are very wise and I am grateful, do any of these options allow me to leave the chip as it is or do I have to upgrade the chip which means I need to go through Fry's electronics and purchase another chip ? or are the options you gave me things which will upgrade existing board to 1.1 grbl,

StuartB4, 2020-01-08:
In the picture you uploaded I think the main chip that stores GRBL is the one I've circled in white underneath the wires.
Cant see it very well but usually they have something like Atmel328 or similar written on them. 
Unfortunately it's soldered to the board and the only way to change it would be to change the whole board. But as I said earlier you can try upgrading grbl thats stored in it using one of the options I mentioned above. The only problem is if you do upgrade it and it does'nt work properly, you might not be able to put it back as it was / is now.  

[image]

quotelawrence, 2020-01-09:
Hi Stuart I used a different board that had a blue nano chip with a reset button on it, it was an extra board I just ordered a little different as for plugs and terminals but after wiring it and loading the newer firmware it say 1.1j so it does have spindle speeds $32= and other settings that the other version .9 did not have but I have to pick up some 2 phase plugs, for spindle motor power and I am still waiting for a replacement laser to come from china I burnt up my 2 phase 2500 mw laser running it at a higher level which I was not sure of the setting I later found out how quickly they burn up thanks again stuart I will probably be quiet until I get the plugs, and the new Laser

StuartB4, 2020-01-09:
Ok.  I usually get a notification email when you post something new.  Good luck with it for now.

quotelawrence, 2020-01-10:
Good morning Stuart I am at work though I am very curious about replacement items there are a lot of plugs 2 pin, 3 pin, 4 pin but I cannot find a parts list if i needed to create jumper leads what are the names or types of contact plugs, up near the ac in put there is a 2 pin plug for spindle power yet this unique plug has a clip that locks it in I cannot find any description also there are the 2, 3, 4, pin connectors all around the board with no reference of size, type, or model is there a spec,  BOM, available that would describe this information

StuartB4, 2020-01-10:
I think the ones you want are called JST connectors.
Like these.
https://www.ebay.co.uk/itm/250x-JST-XH-Kit-2-3-4-Pin-Connector-Plug-Terminal-Socket-Header-Wire-Adaptor/143449643278?hash=item216643d90e:g:7YcAAOSwJwFd2AjA

quotelawrence, 2020-01-14:
yes those looked exact I wanted to point out the disappointment of both GearBest & Banggood companies they either sell you counterfeit products or the products are packaged so poorly they are damaged by the time they arrive my new laser module the adjustable lens fitting was bent as if it had hit the floor packaging was poor no foam or bubble wrap just stuffed in a small tight box, and shipped from china by clumsy lazy shippers

StuartB4, 2020-01-14:
Yeah I know what you mean. I've had quite a few things poorly packaged or damaged. I also noticed a lot of the things have no instructions either, or they do but are poorly translated from chinese language.

closed 2020-01-27 by arkypita